nophead/NopSCADlib #1: Unit functions

DeadlyDad, 2019-06-21:
I see that you have 

`function inch(x) = x * 25.4;                                                        //! Inch to mm conversion
`
I use

`function inch(x) = x * 25.4;                                                        //! Inch to mm conversion
`
`function foot(x) = x * 25.4 * 12;                                                   //! foot to mm conversion
`
`function yard(x) = x * 25.4 * 12 * 3;                                               //! yard to mm conversion
`
`// (For fractional inches, 'inch(1 + 7/8)' will work as expected.)
`

`function mm(x) = x *  1.0;                                                        //! mm specified
`
`function cm(x)  = x * 10.0;                                                        //! cm to mm conversion
`
`function m(x)  = x * 10.0;                                                         //! m to mm conversion
`

While they aren't strictly necessary for my scripts to work, they make it perfectly clear that a number I'm using is a vector of the specified type rather than a scalar.  May I suggest adding them?  

Does that make sense?

nophead, 2019-06-27:
Sorry for the slow response, I didn't realise I had to watch my own repositories.

Yes it makes sense if you design in imperial units. I do everything in metric and only use inch() for parts that have imperial dimensions. I have no objections to adding them though. I will do that.

nophead, 2019-06-27:
Implemented via https://github.com/nophead/NopSCADlib/commit/e386ce792748bbc33ff191a41c392009f7a7af01

closed 2019-06-27 by nophead