nophead/NopSCADlib #3: Sometimes make_all.py fails to generate images of assemblies

BenBergman, 2019-08-11:
I have noticed that sometimes `make_all.py` fails to generate renders of the assemblies. At first I assumed that I had something bad in my `.scad` files so I did a `git bisect` all the way to the first commit where I was able to generate images (I had generated some in between as well). Every single commit I tested failed to generate images. Then I reset everything back to how it was before the `git bisect` and suddenly I got image generation again. It just happened to me again now and I decided to try cloning my project into a new directory and it is still failing there.

The output status of the command is 1, but I'm not seeing any errors in the out (other than "target missing", but I'm not sure what exactly is missing).

Any troubleshooting suggestions?

BenBergman, 2019-08-11:
And as seems to always be the case for me, I managed to find the solution as soon as I posted the issue. I noticed an `openscad.log` file which at the end had

```
DISPLAY=
Can't create OpenGL OffscreenView. Code: -1.
```

So I guess my terminal got disconnected from the display due to some SSHing or something. After opening a new terminal, images are generating as expected.

closed 2019-08-11 by BenBergman

nophead, 2019-08-11:
Glad you solved it. 

"target missing" isn't an error. It is the reason for running OpenSCAD to regenerate the image. If the image already exists then it looks at the mod times of the dependencies and names the first one it finds that is newer. If it doesn't exist yet it just says "target missing".

BenBergman, 2019-08-11:
Ah, good to know. Thanks!