nophead/NopSCADlib #4: Unable to create belt() with pulley outside the loop.

lostapathy, 2019-08-21:
Using this example code, the first pulley in the path should be "outside" the loop of the belt, while the other 3 are inside the belt.

```include <lib.scad>

path = [
  [0, 0, pulley_pr(GT2x20_toothed_idler)],
  [50, 0, pulley_pr(GT2x20_toothed_idler)],
  [-100, 0, pulley_pr(GT2x20_toothed_idler)],
  [-100, 100, pulley_pr(GT2x20_toothed_idler)],
];

belt(type=GT2x6, points=path);
```
The belt path generated around that pulley is:
[image]
Which puts it on a completely different path than intended on the machine.

I'm not really sure if this is a supported use case, but it's come up in a project I've started working with.  Thanks!

nophead, 2019-08-21:
If you want the pulley to be on the outside you need to specify a negative radius. Making that change causes the belt to self intersect, so I also moved it to 5 in the Y direction. You probably would also want to make it a GT2x20_plain_idler because it is running on the back of the belt.

```
path = [
  [0, 5, -pulley_pr(GT2x20_plain_idler)],
  [50, 0, pulley_pr(GT2x20_toothed_idler)],
  [-100, 0, pulley_pr(GT2x20_toothed_idler)],
  [-100, 100, pulley_pr(GT2x20_toothed_idler)],
];
```
[image]

nophead, 2019-08-21:
I added an explanatory comment in 7c1ff5ecd59129a1ea63251c76022b9f9d64c858

closed 2019-08-21 by nophead