nophead/NopSCADlib #7: [Feature Request] RPI4 & USB-C

TheAifam5, 2019-12-16:
As in the title :)

I'm planning to make RPi4 10x Cluster with PoE, all ready but I'm working on the case/cluster model.

nophead, 2019-12-16:
Yes it is on my to do list. I have to model micro hdmi and USB-C first.

dandancheeseandham, 2020-01-15:
I'll add on the Duet 3 6HC to your todo list too 😄 

- [ ]  Micro HDMI
- [ ]  USB-C
- [ ]  RPI4 - [Board dimensions](https://www.raspberrypi.org/documentation/hardware/raspberrypi/mechanical/rpi_MECH_4b_4p0.pdf)
- [ ]  Duet 3 - [Duet 3 Mainboard Dimensions](https://duet3d.dozuki.com/Wiki/Mounting_and_cooling_the_board#Section_Duet_3_Mainboard_6HC_v0_Num_6) & [GrabCAD STEP model](https://grabcad.com/library/duet-3d-duet-3-3d-printer-controller-1)

I may have to see if I can at least add in the mainboard and screw mounting holes to the pcb vitamin myself. I'm not bothered about the rest.

Thank you for a great library.

TheAifam5, 2020-01-18:
USB-C used by RPI4: https://www.toby.co.uk/uploads/publications/2205.pdf
Micro HDMI used by RPI4: https://www.toby.co.uk/uploads/publications/2223.pdf

I will try to make the RPI4 :) But I can't promise - still beginner in OpenSCAD.

nophead, 2020-01-18:
Yes RPi4 is something I should probably add but generally I add things I
need. I do own an RPI4 but I downloaded a case for it and then bought a
metal case, so I so far I haven't needed an model for it.

The way open source works is that people contribute work they have done out
of interest or need. Not a system where people request work to be done and
people give up their free time to do it.

coreylista, 2020-07-27:
I made a basic RPI4 template from the RPI3 version. Added the Micro HDMI slot as well

`RPI4 = ["RPI4", "Raspberry Pi 4", 85, 56, 1.4, 3, 2.75, 6, "green", false, [[3.5, 3.5], [61.5, 3.5], [61.5, -3.5], [3.5, -3.5]], [
            [32.5,  -3.5,   0,  "2p54header", 20, 2],
            [-6.5,     9,   0,     "usb_Ax2"],
            [-6.5,    27,   0,     "usb_Ax2"],
            [-8.5, 45.75,   0,        "rj45"],
            [  54,     6, -90,        "jack"],
            [11.2,     2, -90,      "usb_uA"],
            [   4,    28,  90,        "flex"],
            [46.5,  11.5, -90,        "flex"],
            [39.5,     4, -90,  "micro_hdmi"],
            [  26,     4, -90,  "micro_hdmi"],
            [42.5,    28,   0,       "-block",  85, 56,   5],
            [  27, -24.6,   0,        "chip",  14, 14,   1],
            [  60, -22.3,   0,        "chip",   9,  9, 0.6],
        [7.75, 28, 180, "-uSD", [12, 11.5, 1.28]]
    ], [": Micro SD card"], [32.5 - 9.5 * 2.54, 52.5 - 1.27, 20, 2]
];`

`hdmi_micro = [ "hdmi_micro", "Micro HDMI socket", 8,  5.9, 4.43,  1.4, 2.3,   3,  0.3 ];`

I believe my measurements are accurate.

nophead, 2020-08-01:
I have added the USB-C, micro HDMI and RPI4, thanks for the help @coreylista and @TheAifam5.

martinbudden, 2020-11-08:
Can this be closed now?

nophead, 2020-11-08:
Yes according to the title but the Duet3 request added by @dandancheeseandham needs to be moved to its own issue.

martinbudden, 2020-11-08:
OK, I've split that out.

closed 2020-11-08 by nophead