nophead/NopSCADlib #21: Inconsistent z centering of objects.

martinbudden, 2020-01-13:
There is some inconsistency in how objects are centered on the z-axis. For example `rod`s and `studding` are centered about the origin, whereas `comp_spring`s are drawn starting at `z=0`.

This is just the first instance I have noticed, I imagine there is more inconsistency. It would be good to at the very least identify and list the inconsistencies, but hopefully they could be corrected before the library becomes too widely used.

It would also be good to consider implementing a `center` parameter (like OpenSCAD does) for some if not all objects.

nophead, 2020-01-13:
Possibly springs should have a center option, my threads do.

I tend to default them in the way that is most useful for me. I have never needed a centred spring.

martinbudden, 2020-01-13:
Well it really depends how on how you view the library. If you view it as something of yours that you have made public because you hope other people may find it useful, then it doesn't really matter. If you regard it as something that you'd like to become widely adopted, then consistency is important.

"I have never needed a centred spring." Maybe not. But people will use the library in ways you haven't anticipated. Certainly people will use rotated springs, and one of the advantages of centered parts is that they are easy to rotate.

One thing that makes a library easy to use is the consistency of its interface and the minimisation of unessential complexity. I know springs are not centered and rods are, but what about (say) buttons and switches - I couldn't tell you without trying it out. It's a bit of a pain if for every new part I use I have to work out how to position it.

nophead, 2020-01-13:
Buttons and switches have their origin on the panel they mount to. Similarly stepper motors, etc. It doesn't make sense to centre them on their Z axis as it would be an arbitrary point and useless for mounting. Similarly screws and nuts have the origin where the head sits in the substrate.

Rods, springs and bearings are symmetrical, so it is bit ambiguous where the origin should be. So perhaps they should all have center parameters. They all have a length functions, so it isn't hard to place them without.

I don't have any world domination plans. I mainly published the library so I can publish designs that use it.

martinbudden, 2020-01-13:
I agree that it is logical the way things are done, and I can have a good guess as to how things are positioned, but I can't be sure until I try them out. And yes, it's not difficult to position things either, but all these little things add up in terms of time it takes to do things.

Also, I think I should say, that despite a few seemingly negative comment, I do think this library is brilliant and I enjoy using it.

nophead, 2020-01-14:
I could document the origins of each object. In most cases I think it is
the only logical place. Springs are the odd one out because I think all
other symmetrical objects have it in the centre. Springs are
symmetrical unless they are tapered and I use tapered springs for battery
contacts hence why the origin is at the bottom.

martinbudden, 2020-02-23:
Well, now that I've been using the library a bit more, I don't think this is really a problem. Things are pretty consistent, but I was a bit surprised once or twice as a new user of the library.

Having said that, it would be good to have a center parameter for the spring - I have used springs just once, in an extruder drive block vitamin, and in that case I had to offset by half the length of the spring.

nophead, 2020-02-23:
Yes I can do that. I have also added center options to rods, studding and leadscrews as I was adding threads.

nophead, 2020-02-27:
I have added a center option to spring, which defaults to false which I can justify because some springs are not symmetrical.

I noticed extrusions have no center option so I added one, defaulted to false for backwards compatibility but I think for consistency it should really default to true like rods, studding and leadscrews. What do you think?

martinbudden, 2020-02-27:
Yes, I'm in favour of consistency, so defaulting it to true is fine by me.

nophead, 2020-02-27:
OK I think this can be closed now if you agree.

martinbudden, 2020-02-27:
I agree, this can be closed.

closed 2020-02-27 by nophead