nophead/NopSCADlib #29: non-standard markdown

SmoothieAq, 2020-02-16:
The generated markdown does not show nicely in all viewer.

Tables should have at least 3 dashes, eg not `|:--|` but better `|:---|`.

Would you like a PR? I have also made a carbon tube, would you like PR for stuff like that?

nophead, 2020-02-16:
Yes please. Which viewer doesn't like it?

SmoothieAq, 2020-02-19:
I use IntelliJ for OpenSCAD development, and its markdown viewer is quite strict.

In general I find that https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet has only examples that works in all viewers.

I currently work on some new vitamins, and will make a couple of PRs later; might take a week or two.

nophead, 2020-02-27:
I have fixed this via https://github.com/nophead/NopSCADlib/commit/b559319b0337ed0f1c1e0faa432bf84277e16ecc

closed 2020-02-27 by nophead