nophead/NopSCADlib #30: include missing in nuts.scad

SmoothieAq, 2020-02-16:
nuts.scad is missing an include of washers.scad

nophead, 2020-02-22:
True but it also needed by screws.scad and then it would be include twice.  Unfortunately I don't see anyway to guard against that as would be done in C++ for example. So no vitamin files include prerequisites.  They just have to be included in the right order, or just include lib.scad.

SmoothieAq, 2020-02-23:
Ok, got it :-) Must admit, that I usually myself allows multiple includes to make the prerequisites clear. Your library is consistent and clear in its (other) style, and I am going to just follow that.
You could consider adding some of the missing use's, for example inserts.scad could do with a use of screw.scad :-)

nophead, 2020-02-23:
I think it would slow things down a lot because all the common files would get included over and over again.

nophead, 2020-02-23:
Having said that I now realise I have include some vitamins in other vitamins myself so there is no consistency. I am minded to remove those to avoid them being included over and over again.

The easiest way to use the library is to simple include lib.scad that includes all the rest in the magic order. Alternatively one can look at the test example to see what the prerequisites are.

SmoothieAq, 2020-02-23:
I think the most important thing is to be consistent. I did figure out how to use it, but just hadn't gotten the idea before your answer.
I think it is good to remove all includes from all vitamins. I do think you could add the needed use with no problem. Personally I am not a fan of the big include lib.scad, but it works fine without it :-)

nophead, 2020-02-23:
True an extra use shouldn't be a problem as they are cached.

I use lib.scad on simple one file projects as it doesn't take very long to process it. On Bigger multi-file projects I try to reduce the dependencies as much as possible an just include core.scad and any vitamins I need.

nophead, 2020-02-26:
So I solved this issue by moving washers.scad from screws.scad to nuts.scad.

I also moved screw_insert() from inserts.scad to screw.scad, negating the need to to use screw.scad there. It also means nearly all includes of inserts.scad can be changed to use insert.scad instead because inserts are generally deduced from the screw rather than named directly.

Psus.scad and pcbs.scad tend to use an increasing list of other vitamins, so it would be a pain if they didn't include them. I left those alone, instead I removed linear_bearings, iecs, rockers and microswitches from lib.scad as they get included once by other vitamins.

closed 2020-02-26 by nophead