nophead/NopSCADlib #46: KP08 Pillow Block Bearings Sizes

Drachenkaetzchen, 2020-02-26:
Hi,

due to a measurement mismatch I investigated why the KP08 pillow block was 3mm too high.

Seems that KP08 is specified differently depending on which one you buy. Sometimes the distance from the bottom to the center of the hole is 15mm, sometimes it is 18mm.

I think the "problem" should be documented in the README file, and perhaps a variant of the KP08 pillow block to reflect the height (for example: KP08-15 and KP08-18).

As I don't know which of these is standard, or even if there's a standard, I didn't submit a PR yet.

martinbudden, 2020-02-26:
Interesting. It does seem there are two variants of the KP08 pillow block. I think we should have both variants with names KP08-15 and KP08-18 as you suggest. Would you care to submit a PR?

nophead, 2020-02-26:
Fixed by #47

closed 2020-02-26 by nophead