nophead/NopSCADlib #53: "list index out of range" error when running views.py

hotwolf, 2020-02-27:
Thanks for providing this great library!
I'm using it for the first time and I believe my design follows the documented methodology. However I'm stuck with a run-time error, when I try to execute make_all.py:

```
$ make_all.py 
openscad -D $bom=2 -D $preview=true -o openscad.echo scad/bom.scad 
Generating bom ... done
Traceback (most recent call last):
  File "...library path.../NopSCADlib/scripts/make_all.py", line 39, in <module>
    views(target)
  File "...library path.../NopSCADlib/scripts/views.py", line 111, in views
    assemblies = bom_to_assemblies(bom_dir, bounds_map)
  File "...library path.../NopSCADlib/scripts/views.py", line 73, in bom_to_assemblies
    ass =  flat_bom[-1]
IndexError: list index out of range

```
The design I'm running this on is can be found the following Github repository:
https://github.com/hotwolf/DIYLaserBed

Also, the bom.py call in the beginning generates an empty BOM file.

I'd appreciate it if you could give me a hint, why this is not working or what I'm doing wrong.

Thanks!

nophead, 2020-02-27:
The error is caused by the BOM being empty. I think that is because you have a main_assembly() module but it doesn't call assembly("main"),  it calls assembly("Finished") instead.

All assemblies that are to appear on the BOM should call assembly() with its own name minus the assembly suffix.

I also noticed you include NopSCADlib/lib.scad and then include core.scad and some vitamins. If you include lib.scad you get the core and all the vitamins. The only other includes and uses you may need in addition are just for printed parts and assemblies like boxes.

martinbudden, 2020-02-27:
This is something that caught me out when I first started using the library. The module names and the `assembly` and `stl` parameters must match. So if you have a module called `DIYLB_Top_assembly` then it must use `assembly("DIYLB_Top")`, similarly a module called `DIYLB_CBearing_stl` must call `stl("DIYLB_CBearing")`.

nophead, 2020-02-28:
I added a guard for an empty BOM, so it should at least no fall over at that line now.

hotwolf, 2020-02-28:
Thanks for the fast responses!
I've cleaned up the includes in my design, changed the assembly() and stl() calls, and fetched the latest version of the NopSCADlib. But when I run make_all.py, the generated BOM file is still empty and I get a similar error message:
```
$ make_all.py 
openscad -D $bom=2 -D $preview=true -o openscad.echo scad/bom.scad 
Generating bom ... done
Traceback (most recent call last):
  File "...library path.../NopSCADlib/scripts/make_all.py", line 39, in <module>
    views(target)
  File "...library path.../NopSCADlib/scripts/views.py", line 210, in views
    print('![Main Assembly](assemblies/%s.png)\n' % flat_bom[-1]["name"].replace('_assembly', '_assembled'), file = doc_file)
IndexError: list index out of range

```
Do you have any suggestions what else needs to be fixed?

Thanks!

nophead, 2020-02-29:
Ha, yes I thought it would fall over somewhere else with an empty BOM. I am not sure it is worth fixing views.py to cope with an empty BOM as any project is supposed to contain at least a main assembly. See https://github.com/nophead/NopSCADlib/blob/master/docs/usage.md.

Is there anything in openscad.echo and openscad.log?

hotwolf, 2020-02-29:
The problem is that I'm getting errors when I'm including the latest version of the NopSCADlib, and my design doesn't even load anymore. In my Linux setup (OpenSCAD version 2015.03-2) in which I'm trying to run the Python scripts, I get only get parser error with a high line number without reference to the include file that caused the problem. In my Widows setup (OpenSCAD version 2019.05) my design loads partially and the first warnings say that some types (e.g. "GT2x6", "M3_grub_screw") are now undefined. 
I'll let you know if I figure out which include file causes the problem.

nophead, 2020-02-29:
The first line [here](https://github.com/nophead/NopSCADlib/blob/master/docs/usage.md) says you need at least version 2019.05. The is no way it will work with a version from four years earlier.

hotwolf, 2020-02-29:
I was using the latest version that was available via the Ubuntu PPA, but now I've switched to the latest OpenSCAD AppImage (2019.05). I've been trying to debug my design from the beginning, starting with my own vitamins and the first assemblies. My base assembly still compiles, but with the latest NopSCADlib, all of the extrusions are now centered (pls see [screenshot](https://github.com/hotwolf/DIYLaserBed/blob/master/doc/misaligned_extrusions.png))
Is this that a bug or the new behaviour of the library?

nophead, 2020-02-29:
Extrusions are relatively new to the library and briefly defaulted to not being centred. In the last few days a center parameter has been added and its default changed to true to match a convention in the library that symmetric parts like rods and studding default to being centred. They all now have a centre parameter that can be set to false.

The library does change frequently but usually in backwards compatible ways but not always. If you watch this repository you would see emails whenever there is a change, or you can look at the list of commits.

martinbudden, 2020-02-29:
Once you've got to the stage of having warnings such as `M3_grub_screw` not defined, it is just a question of getting your includes sorted out. Unfortunately OpenSCAD does not have include guards like C, so, in the interests of efficiency, NopSCADlib doesn't include everything needed.

`GT2x6` and `M3_grub_screw` undeined probably means you need to include `belts.scad` and `screws.scad` before including `pulleys.scad`

nophead, 2020-02-29:
Yes it is much easier to start by including lib.scad as that has all the vitamins in an order known to work. 

The alternative is to include core.scad and set OpenSCAD to stop at the first error and add things as needed. Virtually everything needs screws.scad, which also includes nuts and washers.

I would also always load the full design in the GUI first before trying run make_all.py or any of the other scripts. Not sure why you didn't see errors when running bom.py as it is supposed to capture the output and print any errors or warnings.

nophead, 2020-02-29:
I have now included screws.scad in core.scad, so it is no longer necessary to include it anywhere else.

I think all the other vitamins now include all their pre-requisites. I don't think this doesn't create any duplicate includes because I also remove them from lib.scad.

The end result is no hopefully no duplicated includes and no missing dependencies. All vitamins should be included-able on their own after including core.scad.

The original version of core.scad without fasteners is now utils/core/core.scad and is used internally in the library where fasteners are not required.

martinbudden, 2020-02-29:
I think this is a good improvement. Practically everything included screws.scad, and this makes it much easier for new users.

hotwolf, 2020-02-29:
Thanks everyone for your help! 
I've now cleaned up my design to work with current version of the NopSCADlib. It compiles in the GUI again without errors and warnings and I'm now able to generate a BOM. But, make_all.py still doesn't run all the way through:

```
$ make_all.py 
openscad -D $bom=2 -D $preview=true -o openscad.echo scad/bom.scad 
Generating bom ... done
  1.4   0.0 DIYLB_Rail.stl
  2.0   0.0 DIYLB_UBearing.stl
  2.1   0.0 DIYLB_LBearing.stl
  2.1   0.0 DIYLB_CBearing.stl
  6.9   0.0 DIYLB_Carriage.stl
 14.5
openscad --colorscheme=Nature --projection=p --imgsize=4096,4096 --camera=0,0,0,70,0,315,500 --preview --autocenter --viewall -o tmp.png png.scad 
magick tmp.png -trim -resize 280x280 -background #F8F8F8 -gravity Center -extent 280x280 -bordercolor #F8F8F8 -border 10 tmp.png 
Traceback (most recent call last):
  File "...library path.../NopSCADlib/scripts/make_all.py", line 36, in <module>
    render(target, part)
  File "...library path.../NopSCADlib/scripts/render.py", line 68, in render
    % (background, background, tmp_name)).split())
  File "...library path.../NopSCADlib/scripts/tests.py", line 50, in do_cmd
    return subprocess.call(cmd, stdout = output, stderr = output)
  File "/usr/lib/python2.7/subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

```
Do you have an idea what could be the problem now?

nophead, 2020-02-29:
Looks like it can't find Image Magic. Have you installed it. Is it in your path?

hotwolf, 2020-02-29:
Thanks for the hint. My Linux distribution only comes with ImageMagick 6.9.7-4, but I guess NopSCADlib requires ImageMagick 7. I'll give the lastest version of ImageMagick a try.

hotwolf, 2020-03-01:
It worked! I was able to generate my first [assembly instructions](https://github.com/hotwolf/DIYLaserBed). Thanks again for your help.

I have one more question, though: 
In my design, I'm mirroring a full assembly, which is then not considered in the [list of STLs (DIYLB_Carriage)](https://github.com/hotwolf/DIYLaserBed#3d-printed-parts-1). What would be the best way to handle this without duplicating the assembly or at least without showing the assembly step twice in the instructions?

nophead, 2020-03-01:
I would pass a flag to the module that makes the STL change its name. E.g.
```
module my_part__L stl_(right = false) {
   stl(right ? "my_part_R" : "my_part_L");
}
```

Then I would make the mirror version:
```
module my_part_R_stl() mirror([1,0,0) my_part_L_stl(true);
```

I would do the same thing to the assembly to make the right and mirrored left and I would pass the right flag to the call to show the STL.

That would make the BOM correct, there will be right and left STLs and right and left assemblies but they will both be shown in the instructions and you would need to duplicate the instructions or for the second one just say "Assemble as the my_L_assembly() but mirrored".

If you didn't change the name of the assembly when passed the right = true then it would count two of the same assembly and you would only get one set of instructions with twice as many vitamins but with two different STL listed. As long as the instructions said repeat the steps with the right hand version of the STL I think it would make sense. The only problem would be the next level up would use two of the assemblies and show just the left version in the parts list.

nophead, 2020-03-02:
I found that bom.py wasn't showing errors and warnings, so I fixed it with https://github.com/nophead/NopSCADlib/commit/9fbdb8299188673ed1ab477cd19ec791e19dcce1.

I think that was the root cause of this issue and @hotwolf now has it working, so I will close this.

closed 2020-03-02 by nophead

hotwolf, 2020-03-03:
Thanks for the tips on handling mirrored 3D-prints and yes, all my library problems have been resolved.