nophead/NopSCADlib #57: Build scripts assume OPENSCADPATH contains only a single directory

martinbudden, 2020-02-29:
OpenSCAD supports multiple directories (separated by `;`) in OPENSCADPATH, see https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Libraries . However, it seems, `exports.py` and `views.py` assume that OPENSCAPPATH only contains a single directory.

nophead, 2020-02-29:
Perhaps it should locate itself relative to the path of the scripts instead.

martinbudden, 2020-02-29:
Actually, it would be very handy if it searched multiple directories. That way I could have a separate library of printed parts, or use other libraries alongside NopSCADlib.

I don't think it should be difficult to do, just split OPENSCADPATH on `;` and iterate over those paths. This will find `NopSCADlib/printed` and any other libraries with a `printed` subdirectory.

nophead, 2020-02-29:
But there would be no guarantee a directory called printed would contain anything compatible with NopSCADlib.

martinbudden, 2020-02-29:
True. However if NopSCADlib searches the path and finds no `_stl` modules, then no harm is done. The only time anything would actually break is if the user has a non-NopSCADlib compatible`.scad` file with a module with a `_stl` suffix in the `printed` directory. Unlikely, especially since the user is an active user of NopSCADlib.

nophead, 2020-02-29:
True but I have never used anybody else's library for OpenSCAD.

martinbudden, 2020-02-29:
Yes, but you are the owner - if you make a part that is useful to you, you are likely to put it in the library. I have my own library of printed parts - things that are useful to me, but not really generally useful enough for me to justify submitting them to NopSCADlib. I imagine other people are in my situation. This change would allow them to much more easily use their own libraries in a project using NopSCADlib - that, I think, is the real use case.

nophead, 2020-03-01:
I  think #60 will do what you want but it needs testing with multiple libraries in multiple library folders and on Windows and Linux as the path separator changes.

martinbudden, 2020-03-01:
Yes, this looks the business. I can easily test on Windows, and I've got an Ubuntu shell on Windows and I can try and get OpenSCAD running in that.

martinbudden, 2020-03-02:
So I tried to get OpenSCAD running in Ubuntu. It's not immediately available, but I managed to find an installation, so I installed that, but it turned out to be the 2015 version. So rather than go down the rabbit hole of chasing an Ubuntu installation I instead focused on Windows.

I've done quite a bit of testing of this in Windows, moving my library and NopSCADlib to various different directory combinations, and also having non-NopSCADlib subdirectories on the path and it all worked fine. Given you are using python functions to parse the directory and path, I think this is OK to merge.

nophead, 2020-03-02:
OK thanks for testing it. I think the only difference with Unix is the path separator is : instead of ; That should be taken care of as I use os.pathsep

To run on Ubuntu I just download the appimage and do chmod +x on it. I don't have a Ubuntu box in my winter residence, just a Laptop with Win7.

closed 2020-03-02 by nophead