nophead/NopSCADlib #58: There should be an option to turn off magick compare in builds, to speed them up

martinbudden, 2020-02-29:
While useful, the diff files produced by magick compare are not always required. Sometimes it would be better to have a faster build.

nophead, 2020-02-29:
I don't think the diff takes any time compared to the OpenSCAD execution. 

Also the diff is needed to check if the file has changed significantly and to not touch the png if it hasn't. This is needed because building on different machines creates slightly different results. The jaggies in a diagonal edge are in different positions.

It is important not change the png because they contain a timestamp which would make git think they have all changed even if they were identical pixel wise.

martinbudden, 2020-02-29:
OK, I didn't realise that. Closing the issue.

closed 2020-02-29 by martinbudden

nophead, 2020-02-29:
Turning threads off should speed it up!