nophead/NopSCADlib #62: Title-isation of assembly headings is unnecessarily restrictive.

martinbudden, 2020-03-01:
Assembly titles are formed by splitting the assembly function name on underscores and then title-ising the result. So, say, `PSU_cover_assembly` becomes "Psu Cover". This is unnecessarily restrictive to the user, they may prefer "PSU Cover" or "PSU cover". Also title-isation doesn't really confer any benefit - the user could manually get the same title by choice of assembly function name. I suggest the title-isation is removed.

nophead, 2020-03-01:
It is necessary in English I think to start a title with a capital letter but I suppose it should never change upper to lower.

martinbudden, 2020-03-01:
Well:
1. The assembly instructions might not be in English.
2. It is not the job of the library to enforce the rules of English in the assembly instructions - the library should be "policy free". Users can run spell/grammar checkers on the output if they wish.

nophead, 2020-03-01:
I think titles beginning with capitals is universal.

All my assemblies are all lower case. I prefer my assembly instructions to have capitalised titles.

nophead, 2020-03-01:
I have a change that leaves capitals alone but it makes no difference to my build instructions because I don't have any upper case. The only things that look daft are Psu and Xy. I can fix the XY but to fix PSU I would need to change the psu_shroud_assembly in the library.

nophead, 2020-03-02:
A possible compromise is to change the module search to be case insensitive so you can put capitals in the string passed to module() but keep the module names lower case. That way I can fix SSR and PSU in the library but keep the module names the same for backwards compatibility.

If any letter is a capital I can skip capitalisation. That way you would have complete control of the titles other than they can't be all lower case. By default nothing changes and titles are capitalised as before until the module strings are changed to include capitals.

martinbudden, 2020-03-02:
Hmm, I always get a bit wary when things start to get complicated like this...

The essential problem is that title is derived from the module name, and that places restrictions on both.

Ideally I think it would be good to add a second optional parameter to the `assembly` module, it the user would call `assembly(module_name, title);`, where `module_name` is as current, and `title` is an optional title - if ommited, current rules would apply. So you could call `assembly("psu_shroud", "PSU shroud")` or even something like `assembly("bbox", "Butt box")`.

However, I'm not sure if this would be easy to implement, given some of this is done in OpenSCAD, and some done in the python scripts.

nophead, 2020-03-02:
Yes the OpenSCAD only creates the BOM and then the Python script finds the module to match the BOM entry. If they are to be different arbitrarily the BOM line would need to contain both names but they would only need to differ in case to solve this problem in a backwards compatible way. 

That is why I think a case insensitive match is a better solution. Then there is no need to create upper case assembly names at all, although they would still work. 

It would only fail if somebody had assembly module names that only differed in case, but that would be perverse I think.

martinbudden, 2020-03-02:
OK, let's go with the case insensitive match then.

nophead, 2020-03-04:
I have implemented the case insensitive match. It still capitalises each word because there has to be a convention for the assembly titles that come from the library to not be different from the users choice.  I think capitalising each word is a common format for titles, especially those that are not underlined, to stop them looking like a sentence.

martinbudden, 2020-03-09:
I think this can be closed now.

closed 2020-03-09 by nophead