nophead/NopSCADlib #64: SCS bearing block anomaly?

nophead, 2020-03-03:
From this image: 

[image]

It looks like the bearings in the long versions are recessed a lot further than the short versions but pictures on the web don't look like that https://www.makeralot.com/scs12luu-linear-slide-block-p8/

I checked the dimensions of the SCS12LUU and LM12LUU bearings and they both seem correct, for the short version I get 3mm each end but for the long version I get 6.5mm each end. So from the dimensions our picture is correct but it doesn't seem to match reality. I also can't see why the recess would need to be any deeper for a longer bearing.

Does anybody have any of these parts to check. Is the embedded bearing actually longer than a normal long bearing?

martinbudden, 2020-03-06:
Unfortunately I only have size 8 and 12 bearings, and size 8 bearing blocks to check. I took the dimensions from https://www.makeralot.com/scs20uu-linear-ball-bearing-block-p11/ and was careful in doing so, but it would be worthwhile for them to be double checked by someone else.

It seems unlikely that an embedded bearing is actually longer than a normal long bearing.

nophead, 2020-03-06:
It needs somebody with a long version to confirm the recess in each is really 6.5mm ignoring the circlip.

I think the answer might be that the circlip is twice as thick on long versions.

nophead, 2020-03-17:
Another oddity is that a standard DIN472 circlip for a 12mm bore will not clear a 6mm shaft when snapped into its specified groove but all the other sizes would work. Either the groove is deeper than standard or they use a different clip.

nophead, 2020-03-24:
Well the answer to the first anomaly is simple but surprising:

[image]

The long versions don't use the long bearings! They use two short ones and a strange piece of packing 4mm thick in between. The slot for the circlip is much wider than it should be, so they are sloppy when the bearings are not pressing on them.

The C shaped packing doesn't go all the way round as if it is avoiding the tapped M6 hole in the side. Does anybody know what that is for? I thought it might be for a grease nipple but it would not get past the seals of the bearing. It would get onto the rod though.

nophead, 2020-03-24:
BTW, not a good idea to disassemble one as the packing is so hard I think a press is needed to re-seat the second circlip.

nophead, 2020-03-24:
Fixed long SCS bearings to use two short linear bearings separated by 4mm.

closed 2020-03-24 by nophead

martinbudden, 2020-03-28:
Interesting. That certainly explains things.