nophead/NopSCADlib #65: Lacking documentation on how to use BOM, STL and other tools

FLamparski, 2020-03-08:
The scripts for generating BOMs and other documentation items are great, however they are lacking documentation. It would be good to improve the documentation for those tools, because it would make creating projects with nopscadlib a pure joy.

Some items to consider:
* `bom.py`, `make_all.py` and others don't really have a usage doc/man page
* There are some undocumented constraints (eg. for generating STLs, your files must either be under `./scad` or `./scad/printed`; this constraint should also be addressed at some point)
* There is no documentation on how to set up a project in a way that would make the library and tools work best
* Certain dependencies are not obvious (python dependencies not listed in scripts/readme.md, dependency on image-magick CLI tools also not obvious)

nophead, 2020-03-08:
See https://github.com/nophead/NopSCADlib/blob/master/docs/usage.md, which is linked on the main page. That states the files should be in ```./scad```. 

```./scad/printed``` was added recently, so perhaps should be mentioned, but I don't structure projects like that myself.

nophead, 2020-03-09:
Is there anything still not clear after reading usage.md?

FLamparski, 2020-03-09:
It certainly elucidated some things but I'd still like to see usage messages printed by the tools themselves. And for newcomers to the library, putting up step-by-step instructions on how to set up a project could be useful. Finally, aside from the `/printed` directory, it seems that I can't have any subdirectories in my `/scad` directory - perhaps this is worth a mention too, or otherwise a change to the scripts so that they parse the `use` and `include` directives to find files instead of assuming a specific directory layout

nophead, 2020-03-09:
Yes I only use one directory for each project. The biggest has 28 files, most only have between one and three, so I have never considered searching anywhere else except NopSCADlib/printed. 

@martinbudden added ./printed but since my projects mostly generate printed parts with all the vitamins coming from the library, I have never needed a printed subdirectory as all my files would be there.

What things would you separate out into different directories?

nophead, 2020-03-11:
I have added usage messages to all the scripts with 23a64f238d24adfdedbefbc8931e1a82102be219

FLamparski, 2020-03-12:
That looks good, thanks!

nophead, 2020-03-12:
I have also now made it search the dependencies, so the only requirement is it can find the main_assembly in the project's scad directory. Other project directories will then be found and searched. External directories are classed as libraries and will only be searched if they end in /printed. See 2210396234dc9342dbb6c874a4bba4d6f4ba291c

FLamparski, 2020-03-14:
That's pretty awesome, thank you! It should make the library a bit easier to use. I'm gonna close this ticket now but I'm sure that there are many more developer experience tweaks that could be done here.

closed 2020-03-14 by FLamparski