nophead/NopSCADlib #68: PNG colour hard coded to [0, 146/255, 0]

martinbudden, 2020-03-28:
The colour of PNG files is hardcoded to [0, 146/255, 0] in `render.py`. It would be useful if the user could override it so that it was the same as `pp1_colour`, perhaps by a command line option.

nophead, 2020-03-28:
I don't see how that helps as it is normally called from make_all.py.

Somehow the colour used in the assembly should be used because parts can be any colour, not just pp1_colour.

martinbudden, 2020-03-28:
I meant as a parameter to `make_all.py`. And yes, if the colour used in the assembly could be used, that would be better.

nophead, 2020-03-28:
I agree but I can't think of a way to do it.

martinbudden, 2020-03-29:
I can't think of a way to do it either, which is why I suggested a command line option, so that users could at least override the hard-coded colour.

nophead, 2020-03-29:
Problem is the command line format is not compatible with argparse, so adding conventional options is a a pain. In normal use there are no parameters unless it is a multi-target build and then only one that has to be a target.

nophead, 2020-04-05:
OK I worked out how to fix it, see 7e0c5fdb6e7da03a42cc81d87639421821e8b13e

The rendered colour will now default to the ```pp1_colour``` in force when the stl is used in an assembly. To use a different colour use ```stl_colour()``` in place of ```color``` to colour the STL when it is used in an assembly. That will also colour the render.

The only limitation is that you can't have two instances of the same STL with different colours. You would have to give it a different name.

closed 2020-04-05 by nophead