nophead/NopSCADlib #69: No way for user to specify if big or small assembly pictures should be used

martinbudden, 2020-03-28:
`views.py` contains some logic to decide if big or small assembly pictures should be used. However this logic is not always correct and there is no means by which the user can override it.

nophead, 2020-03-28:
It tends to go wrong when large vitamins are used with small plastic parts, Or small milled parts are used. I don't have any ideas to fix that or make it user selectable, do you?

martinbudden, 2020-03-29:
Well, I haven't got my head around the python scripts yet, but could it be either as a parameter to the `assembly` module, or as a comment to the module. ie:
```
module PSU_assembly()
assembly("PSU", picture="big") {
```
or
```
module PSU_assembly() //!! picture=big
assembly("PSU") {
```

Also a command line option to force all pictures to be big would be useful for some projects.

nophead, 2020-03-29:
The best solution would be to modify OpenSCAD to add a command line option to output the camera distance.

martinbudden, 2020-03-29:
I thought you could already specify camera parameters to OpenSCAD.

nophead, 2020-03-29:
You can but I pass --viewall and --autocenter, so it calculates the bounding sphere of the assembly and then sets the camera distance to ensure it is all visible no mater how it is rotated. That makes it a bit small, so I crop the empty border and then resize it with magick.

If I knew the bounding sphere, or the camera distance I would have a good metric of how big the assembly is.

martinbudden, 2020-03-29:
I see. So a change to OpenSCAD is required to properly resolve this issue.

nophead, 2020-03-29:
For it to be automatic, yes. I do make the odd change to OpenSCAD but I like the library to be on a released version rather than the bleeding edge. And there is no guarantee a PR would be accepted. I had a huge argument getting $preview merged.

martinbudden, 2020-03-29:
Interesting that it was hard to get $preview merged.

Any thoughts on what, if anything, could be done in the meantime?

nophead, 2020-03-29:
Perhaps your second suggesting might be straightforward to implement, i.e. annotate the module line.  

The problem with the first one is that the information would have to pass through the BOM.

nophead, 2020-04-04:
Actually I implemented it more like your first suggestion. See https://github.com/nophead/NopSCADlib/commit/bc4e18d788a0df773497b20ba958501855cd2a10

closed 2020-04-04 by nophead