nophead/NopSCADlib #71: color vs colour

martinbudden, 2020-03-28:
I hesitate to raise this as an issue, since I generally have a rule not to argue about anything that can easily changed with a script.

NopSCADlib's use of `colour` is inconsistent with OpenSCAD's use of `color`. This can grate, especially in statements like `color(pp1_colour)`, or even `color(colour)`. Also NopSCADlib is not itself consistent, since we have `microswitch_contact_color`, `d_pillar_color` and other `_color` constants.

Note also that NopSCADlib is not consistent in its use of UK English spelling: `center` is used instead of `centre`.

nophead, 2020-03-28:
I use English English but for some strange reason OpenSCAD uses American English even though it was written by Austrians, and that grates on me.

I try to use colour except where OpenSCAD requires color(). Those color constants I think originated from an American contributor to Mendel90. I tended to use inline colour values and he factored them out to a single file. I moved them back to their point of use in NopSCADlib but they should really all be colour for consistency.

I think center is different because it is an OpenSCAD parameter, so has to be center for built-ins and would be inconsistent and error prone if some module took centre and others centre. I don't think color is a parameter in any OpenSCAD functions or modules, so it can always be colour when passed as a parameter in NopSCADlib.

nophead, 2020-03-29:
I fixed the colour inconsistencies. It is now colour everywhere apart from the calls to color().

So I think it is now consistently English except where OpenSCAD forces US spelling.

closed 2020-04-05 by nophead

nophead, 2020-04-05:
Woops, cited the wrong issue but I think this can be closed anyway.

commit 7e0c5fdb (2020-04-05): Renders of STLs are now the correct colour. Fixes #71