nophead/NopSCADlib #73: slim ball bearings (example: 6808)

crteensy, 2020-04-10:
I made a makeshift 6808 bearing using

`ball_bearing(["6808", 40, 52, 7, "black"]);`

but the rings are a oddly wide. I see no way to change them directly - or is there one?

nophead, 2020-04-10:
No the rim is currently a tenth of the diameter but it could easily be made a parameter, or perhaps a function of od - id. The reason it is a function is because it doesn't seem to be specified.

crteensy, 2020-04-10:
I'll think of something based on od, id and width and see if that returns sane results for the existing bearings and my slim one

crteensy, 2020-04-10:
So I guessed a ball size based on bore, diameter and width and adjusted the angle covered by the rings to get a reasonable fit with actual dimensions I looked up for the existing bearings in the lib and the 6808. The results are within 0.6 mm (diameter) of the actual dimensions - but that might be worse for other series and sizes.

before:
[image]

after:
[image]

If you're happy with this result, I'll fork and push.

nophead, 2020-04-10:
The smaller sizes look too thick to me now. I think it is best to make it a parameter and we all measure the rims of the bearings we have. I have the first two and I think @martinbudden contributed the next two. We can fill them in with diameter / 2 until we have measurements.

crteensy, 2020-04-11:
We don't need measurements because the diameters are given when you look up manufacturer data (which I did to calculate the errors introduced by my guessing method mentioned above).Most online stores don't specify these diameters because they're somewhat secondary in hobby use. When you look up a bearing in FAG Schaeffler's catalog, for example, they are part of the dimensions given. However, they also depend on the kind of seal to a certain extent.

A word of caution: When the bearing dimensions vector is enlarged to contain two more dimensions, this might break existing code where people have put in their own bearing (like I did). So I think a fallback for shorter vectors might be useful.

nophead, 2020-04-11:
If people create there own vitamins and don't add them to NopSCADlib with a PR they will generally break because I often add more parameters and then update the object lists in the library. So if you use ball_bearing(BB608) that will always work and is backwards compatible.

I.e. the interface of the library is considered to be the functions, modules and objects names. The contents of the object lists are considered internal, that is why they all have accessor functions. Those isolate changes to the order and contents of the list.

Looking at real bearings, they have chamfers and the outer rim is thinner than the inner one.

Looking at the Schaeffler's catalogue I can't find the bearings with rubber seals I use and they have radii instead of chamfers. I buy my bearings on eBay, so I don't have a manufacturer name.

I will update the model to have two more diameters and perhaps the chamfer.

nophead, 2020-04-11:
Actually looking at pictures of 608-2RS bearings online there seems to be a big variation in the rims and nobody specifies them. If I change the calculation to ```(bb_diameter(type) -  bb_bore(type)) / 7``` it makes only a minute difference to the existing bearings. How does it look for yours?

crteensy, 2020-04-11:
Yes, the exact dimensions vary a bit - that's why I think that a best guess will do the job just fine. How exactly that works, and for which bearings we tune it, is a matter of taste. I'd be happy if the standard bearings look somewhat reasonable and the slim ("deep groove") bearings look less odd.

nophead, 2020-04-11:
I have made the hub and rim widths properties and measured the three bearings I have and guessed the last two. To make it more realistic I have added chamfers because otherwise they look too wide with the correct dimensions.

[image]

crteensy, 2020-04-11:
The chamfers look a bit large, but other than that - nice! You even guesses the rim sizes for the 6808 right (inner is 1.6, outer is 1.5)!

nophead, 2020-04-11:
OK I made the hub exact and reduced the chamfers a bit.

[image]

As a matter of interest what do you use the 6808 bearing for?

crteensy, 2020-04-12:
I use it to rotate a vacuum pickup tool carriage:
[image]
The servo at the top moves the carriage (in the middle) up and down. The carriage holds a syringe with a blunt needle to pickup SMD components, and the whole thing can be rotated. That's the overall idea at least.

crteensy, 2020-04-12:
BB6808 works for me now, thanks a lot!

closed 2020-04-12 by crteensy