nophead/NopSCADlib #74: Merge to BOLTS

KOLANICH, 2020-06-16:
There is a similar project called BOLTS. How about merging to it?

nophead, 2020-06-16:
It doesn't seem compatible in any way to me.

```
ISO4014("M8", 60);
echo(get_dim(ISO4014_dims("M8"),"d1"));
```
[image]

```
screw(M8_hex_screw, 60);
echo(screw_head_radius(M8_hex_screw) * 2);
```

[image]

Every aspect of specifying parts and positioning them is a totally different philosophy.  Also I think it is made from metadata and supports more than one CAD format, so the primary source isn't even OpenSCAD.

KOLANICH, 2020-06-16:
Yeah, there are duplicated parts, but there are also parts present in different places. Anyway, the positioning philosophy is highly object-specific, so we cannot expect that BOLTS/NopSCADlib had qny for object types not present in it. IMHO it is anyway better to have 1 lib rather than 2. Bigger the lib - more people will be involved into maintainment of it.

nophead, 2020-06-16:
The duplication isn't the issue, the two libraries have totally different ways of specifying a part, extracting its properties and positioning it, which is what I tried to show. 

I can't use ISO4014() in my designs because it doesn't use my screw interface. So an object that took a screw as a parameter could be passed M8_hex_screw or M4_cap_screw, which are property lists and draw either with screw() but it can't be passed ISO4014 is a module name. There is currently no way in OpenSCAD to pass a module as a parameter or store it in a property list. So I don't see how you could even have an object that could use different screw types. They would have to use a fixed screw type and just be passed "M8" or "M4" to specify the size.

My screws have their origin under the head and I translate and rotate them into place. BOLTS has some connector system to join things together, which I have never see a need for.

So the only things they have in common is the licence, no way the code can be merged because they are not even in the same language. NopSCADlib is written in OpenSCAD but BOLTS seems to start with metadata that is translated into OpenSCAD, or FreeCAD format.

closed 2020-06-16 by nophead