nophead/NopSCADlib #78: codespell version conflict

BenBergman, 2020-07-24:
I'm getting the following error when it gets to the codespell part of the make script. My python experience is limited, so I'm not sure how to resolve this.

```
codespell -L od readme.md 
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 582, in _build_master
    ws.require(__requires__)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 899, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 790, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.VersionConflict: (codespell 1.17.1 (/home/ben/.local/lib/python3.8/site-packages), Requirement.parse('codespell==1.15.0'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/codespell", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3257, in <module>
    def _initialize_master_working_set():
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3240, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 3269, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 584, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 597, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/lib/python3.8/site-packages/pkg_resources/__init__.py", line 785, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'codespell==1.15.0' distribution was not found and is required by the application
```

My system defaults to python 3, but I've done a `pip2 install codespell` and `pip3 install codespell` to make sure both bases are covered.

nophead, 2020-07-24:
It seems to be saying it wants version 1.15.0 but it is finding 1.17.1 but I have no idea why because I haven't specified the version required anywhere. It seems more like the sort of error you should get installing it, not while running it. Also seems like an issue you should address to codespell as simply installing it and running it seems to fail.

You might be able to fix it by uninstalling one of the two versions with pip3. The version I am running is 1.15.0 but assuming the command line stays the same any should work.

BenBergman, 2020-07-24:
I hadn't realized just running codespell was showing the error, but I can confirm that to be the case. I'm on Linux, so maybe I've got a system wide install that left an artifact on my system. Deleting the link at `/usr/bin/codespell` (which is where I found a version reference) and installing the system package I've got it working now.

Thanks for pointing me at the actual problem and sorry for the noise!

closed 2020-07-24 by BenBergman

nophead, 2020-07-24:
No problem with noise, this is a quite quiet repository!

Good to have a new user.