nophead/NopSCADlib #79: How to export model of final assembly?

BenBergman, 2020-08-12:
I rendered an STL of my full assembly once many months ago but I'm having a hard time figuring out how to do it now. Is this something that is supported by NopSCAD or did I manage to hack something together back then?

nophead, 2020-08-12:
Not it isn't supported, it only makes PNG views of assemblies and individual STL files for the printed parts. Not sure an STL of a final assembly makes much sense as it will usually tax OpenSCAD to the max trying to union all the screws and washers, etc.

To do it you could load the main assembly in the GUI and press F6 and then export the result but I normally guard my main assemblies with if($preview) because if I accidentally hit F6 it will go away for a long time and usually crashes if the assembly is complex.

BenBergman, 2020-08-14:
I think that is what I ended up doing last time. I had tried it a couple times before creating this issue, but one of the STLs I imported as a vitamin was causing issues and crashing the related assemblies, but commenting it out got the rest rendered.

While I generally agree it isn't useful to make a master STL, my use case is a bit unique. I'm designing a custom VR controller and the artist on the project needs a reference for the actual scale of the device for the in game assets. A unified model serves that purpose best.

nophead, 2020-08-14:
Yes I code vitamins to preview quickly and look correct, so they are not necessarily manifold. Also positioning fasteners flush with the mating surface can create cGAL errors if you try to union them together as they might need to overlap slightly. 

If you just need the bounds taking the hull might be a lot faster and more robust but I haven't tried it.

nophead, 2020-11-13:
Closing this because, in general, OpenSCAD will crash making an STL of an assembly and if it doesn't the result will not be manifold. 

It can be done manually with F6 anyway and I think adding an automated way of doing it will just creates lots of reports that it crashes.

closed 2020-11-13 by nophead