nophead/NopSCADlib #80: DXF Example

niclash, 2020-10-27:
I have problem generating DXF for routing and the "box example" doesn't have any `dxf("that")` to look at nor test it on my system, as I suspect that I am missing some dependency somewhere.

nophead, 2020-10-27:
The box example here: https://github.com/nophead/NopSCADlib#Box generates 6 DXF files as listed here: https://github.com/nophead/NopSCADlib#routed

They are generated by lines like this: https://github.com/nophead/NopSCADlib/blob/master/printed/box.scad#L428

niclash, 2020-10-27:
So when I add `dxf("bracket_x1_axis");` to [my code](https://github.com/niclash/DreamPlacer/blob/edc9ec78ca09685f297a0368a1d5d1b0b40665f6/scad/dreamplacer_main.scad#L67), then I get the following error and early termination;

```
openscad-nightly -D $bom=2 -D $preview=true --hardwarnings -o openscad.echo -d bom/bom.deps scad/bom.scad 
Generating bom ... done
dxfs/bracket_x1_axis.dxf missing
openscad-nightly -D$bom=1 -d deps/dreamplacer_main.deps -o dxfs/bracket_x1_axis.dxf dxf.scad 
```

The DXF is generated, right?

nophead, 2020-10-27:
Instead of: 
```
module bracket_x1_axis_dxf() 
xf("bracket_x1_axis") {
{
```

It should be:
```
module bracket_x1_axis_dxf() 
{
   dxf("bracket_x1_axis");
```

I.e. the dxf module does not take any children.

If that doesn't fix it try listing openscad.log to see if it has any error messages.

niclash, 2020-10-27:
Yeah, I know that and you must have managed to load the old commit, before I noticed that I had forgotten to push my local changes. Check again... And the the error message given, was from the  `dxf("bracket_x1_axis");` case (as it was local).

nophead, 2020-10-27:
What is the error message?  I can't see one?

It is normal for it to say the dxf is missing the first time as the reason for running openscad to generate it.

niclash, 2020-10-27:
Oh, ok...  Well, it aborts the processing. What I posted above is all output there is. if I remove the `dxf()`, I get dozens of lines of processing...

nophead, 2020-10-28:
What does openscad.log say?

niclash, 2020-10-28:
Ah!! I should have seen that file...  "Top level object is not a 2D object"
That actually explains a thought I had; "How does it know what plane I am interested in?"

So, I need to draw the polyline in 2D and then extrude it, rather than `union`+`difference` the 3D shapes of `sheet()`, right?

nophead, 2020-10-28:
The fastest way is to model the sheets in 2D using sheet_2D() for the blank and subtracting circles instead of cylinders, etc.

When you assemble the sheet into the assembly you use render_3D_sheet() to extrude and colour it. See https://github.com/nophead/NopSCADlib/blob/master/printed/box.scad#L578

niclash, 2020-10-28:
That assumes a rectangle shape... But anyway; THANKS a lot for your patience and guidance. I started using NopSCADlib because of the massive library and much later discovered the documentation features, which are just GREAT. I think I can manage from here on.

nophead, 2020-10-28:
It only assumes the blank is rectangular. I always start with rectangular sheets but the results can be any shape. I have made circles and hexagons using intersection and sheet_2D. The vast majority of shapes I make are rounded rectangles with cutouts, which is why sheet_2D can specify corner radii.

I noticed you unioned two sheets but that would make two BOM entries and I doubt you would buy two sheet and join them.

niclash, 2020-10-28:
But union/intersecting multiple 2D sheets will also produce multiple BOM entries, IIUIC.

nophead, 2020-10-28:
Each of my DXFs start with just one sheet and then I use difference and intersection with geometric primitives to get the shape I want to cut from it. I never combine sheets as that would imply starting with multiple sheets and somehow joining them together, which I never do.

The BOM entry is then the smallest rectangular sheet the part can be cut from. I can actually buy sheets cut to size with radiused corners but in practice I use blanks 20mm bigger and clamp them around the edges and then mill the full outline.

niclash, 2020-10-28:
Got it! Smart!

Another "issue"; If I follow the pattern in https://github.com/nophead/NopSCADlib/blob/master/printed/box.scad, I get that the "xyz_dxf" module is missing. But in the box.scad, there is also no (e.g) "box_front_dxf" module. Is the box.scad some re-written/regenerated code from another source?

nophead, 2020-10-28:
The user of the box has to write those because they need to be customised. box_assmbly.scad is included in the user's code and some modules are redefined like this:
```
module box_front_dxf() {
    difference() {
        box_front_blank(box);

        translate([socket_x, socket_z])
            mains_socket_holes(socket);

        translate([meter_x, meter_z])
            panel_meter_cutout(meter);

        translate([switch_x, switch_z])
            rocker_hole(switch, 0);

        translate([neon_x, neon_z])
            rocker_hole(neon, 0);

        for(x = [jack_x1, jack_x2, jack_x3])
            translate([x, jack_z])
                drill(jack_4mm_shielded_hole_radius(), 0);
    }
}

module box_front(type)
assembly("box_front") {

    render_2D_sheet(box_sheets(box)) box_front_dxf();

    translate([socket_x, socket_z, side_sheet_thickness / 2])
       socket_box_fastened_assembly(socket, side_sheet_thickness);

    translate([meter_x, meter_z, side_sheet_thickness / 2])
        explode(30)
            panel_meter(meter);

    translate([switch_x, switch_z, side_sheet_thickness / 2])
        rocker(switch);

    translate([neon_x, neon_z, side_sheet_thickness / 2])
        rocker(neon);

    for(i = [0 : 2])
        translate([[jack_x1, jack_x2, jack_x3][i], jack_z, side_sheet_thickness / 2])
            jack_4mm_shielded(["sienna", "dimgrey", "royalblue"][i], side_sheet_thickness);
}

module main_assembly()
assembly("main") {

    box_assembly();
}

module bottom_bezel_stl() box_bezel(box, true);
module top_bezel_stl() box_bezel(box, false);
module box_corner_profile_stl() box_corner_profile(box);
```

niclash, 2020-10-28:
Ah! Got it... And got it working on my end.

niclash, 2020-10-28:
First one working; https://github.com/niclash/DreamPlacer#cnc-routed-parts

Q; I assume that "dimensions" are to be done "post-processing" on the DXF. Right?

nophead, 2020-10-28:
OpenSCAD is unitless but by convention 1 unit is 1mm and my CAM tools are set up that way. So I just convert the DXF to gcode with PyCAM and the dimensions are correct.

niclash, 2020-10-28:
Ok... Once again; Superb library!!! And OpenSCAD really suits a programmer like myself ;-)

closed 2020-10-28 by niclash