nophead/NopSCADlib #81: The `tests.py` script is hardcoded to NopSCADlib, making it not generally useful.

martinbudden, 2020-11-08:
I've got a project with some tests in the `tests` directory. `tests.py` cannot currently be used to run them, but a few small changes could rectify this. In particular:

1. `tests.py` does not search the project `tests` directory. Adding `tests/` to the `locations` array fixes this.
2. The output is hardcoded to `readme.md`, which means it overwrites the output from `make_all.py`.
3. `libtest.scad` is required for `tests.py`. This is not required in the general case.
4. The heading blurb is directly encoded in `tests.py`. It would be better if it was scraped from `libtest.scad` - this would have the added benefit of making it easier to maintain the blurb.

nophead, 2020-11-08:
Well yes it is designed specially to test the library and make its documentation. I don't have any tests in my projects other than building them makes their documentation and I see when stls and assembly views change unexpectedly, which is really only what tests.py does for the library.

I don't have anything resembling the library structure in my projects. For example it assumes vitamins, utils and printed directories and it produces a contents section split on three hard coded letters of the alphabet.

I don't have properly coded vitamins in my projects. Most projects only have one file in them and get everything from the library. I do mock up a few vitamins specific to a project but don't bother to parameterise or document them unless I want to use them in more than one project and then I move them to the library and make them more general.

What do you have in you projects that needs testing that isn't covered by make_all.py?

martinbudden, 2020-11-08:
There are a few reasons I have tests that suplement `make_all.py`:

1. `make_all.py` takes ages to run if the project is largish.
2. I do have vitamins in my project. From your perspective, if you create a new vitamin, you might as well put it in the library. From my perspective, the vitamin might not be generally useful, or you might not want it in the library for some reason.Or the vitamin might be "rough and ready", that is its dimensions are correct, but it lacks colouring and decoration and I might not have the time to get it to library standard.  So I have test files to test vitamins. 
3. I can create test files for "unnatural" subassemblies. That is subassemblies that don't actually occur in the project, but that are created to test fit and alignment without other objects getting in the way. Also "naked" subassemblies can be useful, that is subassemblies that don't have the outside walls, or the structural elements (such as extrusions).

I managed to get `tests.py` working for my purposes. There are very few changes required and I think they might be useful to others.

nophead, 2020-11-08:
Yes make_all gets very slow as projects get larger. I am hoping the persistent cache addition to OpenSCAD will help there.

I have no objection to a PR making tests.py more general as long as it still works for the library and is not too hard to maintain.

If you have it working can you make a PR please?

martinbudden, 2020-11-08:
Well, my change hardcodes the output to `tests.md` and `tests.html`, which is probably not what you want for NopSCADlib. But with that proviso, I'm happy to make a PR.

nophead, 2020-11-08:
The file needs to be called readme.md for GitHub, so we need a way of detecting it is being used in a project because in the project the build instructions also need to be readme.md if you want to publish the project on GitHub.

nophead, 2020-11-09:
I have a version working and tested in one of my projects. The output looks like this
[image]

It will work for testing vitamins, utilities and printed parts if you put them in correct folders. I am wondering what to do about your tests of assembles. We could have an assemblies folder or I could look in the scad folder and call it a misc category if it finds the source there. Or I could do both.

martinbudden, 2020-11-09:
Why don't you release it as is, I'll try it out and come back with my suggestions?

nophead, 2020-11-09:
OK have done.

martinbudden, 2020-11-10:
Looks good. If you add:
```
            if os.path.isdir('tests') and not libtest:
                locations.append(('tests/' + base_name + '.scad',                 'Tests'))
```
to the end of where you build up the locations array, it works great for me.

Don't know if that fits in with your usage though.

nophead, 2020-11-10:
No because, like your PR, it doesn't make sense for me. ```tests``` is where the tests are and it runs through all the files there. ```locations``` is where it looks for the source of the things to be tested. When tests is added to that list it always exists, so creates a Tests category in the contents and finds itself as the source code for the thing to be tested. In my project it would create an empty column as I have moved the vitamins I am testing to ```vitamins```, so it finds their source code.

I assume the problem you are trying to solve is that things you are testing are not in their own files with the same name as the test itself, so you would get "Can't find implementation!" and it would skip the test.

If that is the case I think it just needs this:
```
            else:
                if libtest:
                    print("Can't find implementation!")
                    continue
```

nophead, 2020-11-10:
Actually it is more complicated to fix than that.

nophead, 2020-11-10:
OK I pushed a version where you can have a test that doesn't correspond to an implementation file.

[image]

nophead, 2020-11-10:
When it can't find the implementation it scrapes the blurb from the test file. In this case I just instantiated a random assembly in main.scad. 

[image]

martinbudden, 2020-11-10:
Yes, this does the business. Great, thanks.

nophead, 2020-11-10:
Fixed via https://github.com/nophead/NopSCADlib/commit/eac0086199b8726c79230c51550bb019f9d2f5b8

closed 2020-11-10 by nophead