nophead/NopSCADlib #85: Possible error in GT2x16_plain_idler width

martinbudden, 2020-11-08:
In `pulleys.scad` `GT2x16_plain_idler` is specified as 7.0, whereas `GT2x20_toothed_idler`, `GT2x20_plain_idler`, and `GT2x16_toothed_idler` all have a width of 6.5. Additionally my `GT2x16_plain_idler` and `GT2x16_toothed_idler` pulleys have the same width, and I have seen spec's on the web stating it should be 6.5mm.

nophead, 2020-11-08:
It came from RobotDigg and they don't seem to have drawings or dimensions, so I probably measured it. Quite possible I have made a mistake or a typo because I haven't used them yet but I am in my winter residence now and don't have access to the parts until the spring. Where is the spec you are looking at from?

martinbudden, 2020-11-08:
I've had a look at several on aliexpress, eg https://www.aliexpress.com/item/4000183519445.html - you need to scroll down quite a way to get the dimensions.

nophead, 2020-11-08:
On the other hand it is a good match for this https://www.amazon.com/WINSINN-Pulley-Toothless-Aluminum-Printer/dp/B07BPHPYTN?th=1. So it looks like both are available.

martinbudden, 2020-11-09:
It does seem that both sizes are available. However I do assume they are matched, ie a 7mm toothless pulley will have a corresponding 7mm toothed pulley. Designing a belt system with unmatched pulleys is a nightmare. As far as `pulleys.scad` goes, it should have either consistently 6.5mm, consistently 7mm or both. I favour consistently 6.5mm.

nophead, 2020-11-09:
Well there will need to two variants then because it seems that RobotDigg sold me unmatched ones.

martinbudden, 2020-11-09:
That's a nuisance!

nophead, 2020-11-09:
It doesn't make much difference to the design, my plastic parts will just adjust to match.

I will make the default 6.5 and add a 7mm variant.

nophead, 2020-11-13:
Fixed via https://github.com/nophead/NopSCADlib/commit/520569cb30c82be08b2019f6c8e371453c0aeed6

closed 2020-11-13 by nophead