nophead/NopSCADlib #88: Socket depth for M3_grub screw is too large

martinbudden, 2020-11-10:
It is currently 2.5mm, which is more than M3_cap_screw or M3_dome_screw. It looks a bit odd as it is, and also it means there is interference when a grub screw is used without making a corresponding hole in a vitamin. I suggest it is changed to either 1.3mm or 1.0mm.

nophead, 2020-11-10:
It reflects the screws I actually have, like most things in the library and I never place screws where there aren't holes.

The reason they are deeper than cap screws and dome screws is the socket is smaller diameter, so it fits inside the threaded part, so can be as deep as the screw, and in fact you can get grub screws where it goes all the way through. For screws with heads the socket is bigger and has to be contained withing the head.

martinbudden, 2020-11-10:
Well, I sometimes add a part without a hole in a vitamin, on the grounds that drawing will be quicker, but if the library policy is always to have holes, that is fine by me.

Closing.

closed 2020-11-10 by martinbudden