nophead/NopSCADlib #91: Belt gap does not allow a belt to be offset with itself at the gap positon.

martinbudden, 2020-11-13:
This is required, for example in certain forms of coreXY. See image, the current belt gaps are highlighted in pink.
[image]

Fix will be submitted soon.

nophead, 2020-11-13:
Interesting. Is that core XY with two separate planar belts? I thought it had one belt in a spiral normally. I wouldn't like to model that so use H-BOT myself! I don't think the racking force matters with trapezoidal rails. I have a milling machine with a single Y axis ball screw in the middle. I have yet to test my H-BOT though.

martinbudden, 2020-11-13:
Yes, that is coreXY. CoreXY always uses two belts, although there are different ways they can be offset from each other. It's not that difficult to model. Here's the top view with the fix (and you can see why I wanted the JST connectors on the motors).
[image]

martinbudden, 2020-11-13:
And for your interest, here's current state of the project:
[image]

nophead, 2020-11-13:
Not sure what is happening to your latest commit to the PR. I see this:

[image]

If I click on the commit number in the email I get this:

[image]

martinbudden, 2020-11-13:
I think I made a mistake when rebasing. It should be ok now.

nophead, 2020-11-14:
Seems there are two ways of doing coreXY. From the Reprap wiki:

>There is two ways to built the CoreXY belt system.
>
>Both belts are on same plane so belts are crossing and they tend to rub one against the other at crossing point. Twisting the belt on the crossing path limits the rubbing. A small finger can be installed to separate the belts at crossing point.
>The two belts could be set on two different levels (see image). This creates a small tilting momentum on the hotend carriage and increase the momentum on shafts if pulleys are installed on cantilevered shafts. This also save some space on the return pulley side as the pulleys could be stacked.

It would be difficult to model the second type because the path is no longer 2D.

martinbudden, 2020-11-14:
Yes, modelling the second type would be difficult in NopSCADlib.

And among the two types, there are many variations.

closed 2020-11-17 by nophead