nophead/NopSCADlib #93: Suggestions for improvements to main library picture

martinbudden, 2020-11-13:
Some minor changes in the grouping

1. Move variacs and transformers to the right and put the iecs up below the power supplies
2. Move the cable strips to the right and put the blowers to the right of the fans
3. Swap the block containing the extrusion, panel meters and stepper motors with the block containing the bearing blocks, the pillow blocks and the sk_brackets. That groups things that have to do with rods nearer the rods.

nophead, 2020-11-13:
Things are generally ordered so that the smallest / lowest are at the front left corner, so they are easier to see. There is currently no attempt at logical grouping.

martinbudden, 2020-11-13:
OK. Closing.

closed 2020-11-13 by martinbudden