nophead/NopSCADlib #99: make_all.py script does not create all directories required

martinbudden, 2020-11-15:
In particular `stls/printed` and `dxfs/routed` are not created.

nophead, 2020-11-15:
It does for me here: https://github.com/nophead/NopSCADlib/blob/master/scripts/plateup.py#L53

nophead, 2020-11-15:
It wont if  you haven't got any platters or panels.

martinbudden, 2020-11-15:
There's the rub. I don't have any platers or panels, so I don't need `dxfs/routed`, yet `make_all.py` fails because it does not exist.

nophead, 2020-11-15:
There is no point to printed when you haven't got platters and no point to routed when you haven't got panels because they would just be a copy of their parent directories.

I have lots of simple projects that haven't got them. Where is it falling over?

martinbudden, 2020-11-15:
Here's an output from by build:
(BTW, it's copying a file from `stls` to `stls/printed` - why is that?)
```
python ../../openscad/nopscadlib/scripts/make_all.py
openscad -D $bom=2 -D $preview=true --hardwarnings -o openscad.echo -d bom/bom.deps scad/bom.scad
Generating bom ... done
  0.3   0.0 zRodMountGuide_21mm.stl
  0.3   0.0 zMotorMountGuide_54p5mm.stl
  0.5   0.0 standoffTap_20mm.stl
  0.5   0.0 standoff_10mm.stl
  1.8   0.0 endstop.stl
  4.2   0.0 beltAdjuster.stl
  4.3   0.0 beltClamp.stl
  5.4   0.0 BigTreeTechTFT_3_5v3_0_bracket.stl
  5.5   0.0 ExtrusionFootLShaped_12mm.stl
  6.4   0.0 xCarriageBottomLR.stl
  8.1   0.0 spoolHolder.stl
 11.1   0.0 zLeadScrewBearingHolder.stl
 11.6   0.0 PSU_cover.stl
 12.0   0.0 printBedCornerBracket.stl
 14.2   0.0 xyIdlerLeftLR.stl
 18.2   0.0 zCarriageSCSRight.stl
 18.4   0.0 zMotorMount.stl
 19.0   0.0 xyIdlerRightLR.stl
 19.3   0.0 zCarriageSCSLeft.stl
 22.8   0.0 yCarriageRightLR.stl
 24.1   0.0 xyMotorMountLeftLR.stl
 28.1   0.0 yCarriageLeftLR.stl
 30.8   0.0 xyMotorMountRightLR.stl
129.4   0.0 xCarriageLR.stl
396.4
Copying stls/beltAdjuster.stl to stls/printed/beltAdjuster.stl
Traceback (most recent call last):
  File "../../openscad/nopscadlib/scripts/make_all.py", line 45, in <module>
    plateup(target, part)
  File "C:\Users\martin\Code\GitHub\openscad\nopscadlib\scripts\plateup.py", line 100, in plateup
    copyfile(src, dst)
  File "C:\Users\martin\AppData\Local\Programs\Python\Python38\lib\shutil.py", line 259, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: 'stls/printed/beltAdjuster.stl'
```

nophead, 2020-11-15:
Have you got and empty platters directory?

nophead, 2020-11-15:
Hmm, just tried it on a project and it fails for me as well. I will try to work out what broke it.

nophead, 2020-11-15:
I pushed a fix and gave it quick test but have to go out now.

martinbudden, 2020-11-15:
That seems to have fixed it. Thanks.

nophead, 2020-11-16:
Fixed via https://github.com/nophead/NopSCADlib/commit/eb9bcf0adadce8b3ed74bd5bd8e4cc1ceae30169.

Seems that I broke it in June and nobody built a simple project without platters and panels since then including me!

closed 2020-11-16 by nophead