nophead/NopSCADlib #103: [Feature Request] Add support for c-style comments for documentation

martinbudden, 2020-11-22:
That is allow `/*! ... */` comments, so
```
/*!
blurb
blurb
blurb
*/
```
could be used as well as
```
//!blurb
//!blurb
//!blurb
```
This would make editing documentation much easier.

nophead, 2020-11-27:
Problem is the simple parsing code is structured around processing one line at a time. It would need to maintain state between lines and get more complicated.

martinbudden, 2020-12-13:
Agreed it would be slightly more complicated. But the state is a simple one - either you are in a comment or you are not. It would be perfectly fine to force a quite if the user started to nest comments or anything vaguely complicated.

I think it's worth it - this would really make a difference when writing assembly instructions.