nophead/NopSCADlib #105: [Feature request] Add support for build variants

martinbudden, 2020-12-13:
Quite often it is useful to have variants of a project. At a simple level this could an item in different sizes, but items could be customised in various ways.

I suggest something like the following:

1. a new `variant` parameter to the scripts (eg `make_all.py`). This is passed on to OpenSCAD as `$variant` which can be used as desired.
2. when using a variant, the build scripts look for `$variant_assembly` rather than `main_assembly`, ie if the variant was called `small`, then it would look for the `small_assembly()` module.
3. the generated output files are put in a variant subdirectory, eg `small/readme.html`, `small/stls` etc.

Would this be feasible? I'd find it very useful.

nophead, 2020-12-13:
I already have support for build variants. See https://github.com/nophead/NopSCADlib/blob/master/docs/usage.md#multiple-configurations.

I do pass a target name to all the scripts and it does make the files in sub-directories with the same name. I am using it to make multiple sizes of a parametric 3D printer.

Rather than setting a single variable it makes a file called target.scad that includes config_<target>.scad so you can set whatever variables you want in those config files. Once you create files with that name pattern you get an error message if you don't pass a target name to each script. If you don't have them you get an error if you do pass one.

I don't look for different assembly modules though. You could switch on a variable if you need that. I.e. have a main_assembly() that calls a different module depending on variables you see in the config file.

martinbudden, 2020-12-13:
Thanks, I'll have a look at that.

closed 2020-12-13 by martinbudden