nophead/NopSCADlib #106: Suggestion - add option to rotate each layer of a polyhole

martinbudden, 2020-12-15:
Not sure if this is a good idea or not. What about rotating each layer of a polyhole, like is done with the poly countersink. I think there are two possible advantages:

1. Cosmetic. When viewed from above on a printed part, the polyhole will look more circular.
2. It might give a better "grip" when used for self-tapping holes?

What do you think? Is this a good idea or not?

nophead, 2020-12-15:
Interesting idea. When printing ABS most holes look fairly round until they get very small and then I can see an advantage alternating them.

The countersink layers don't line up because the number of facets change. I am thinking I might need to make the number of facets constant when printing inverted and alternating each layer so that the vertices always land well over the edges of the layer below.

Possibly polyholes should alternate by a half angle each layer, certainly when they number of sides gets small. For example when it is triangle I think it would be better to alternate them. The only downside I can see is it increases the number of facets, which will slow down a CGAL difference. I often do the subtraction in 2D and extrude the result. I would have to do it all in 3D to take advantage.

Not sure what you mean by 3. Do you mean a hanging hole? There is a module for that.

I am away from my printers at the moment so can't test anything.

martinbudden, 2020-12-15:
By 3, I mean where you have a bridge on a component, and then later decide there needs to be a vertical hole in that bridge. For M2 holes you can generally get away with this even with a cylinder for the hole. I've just added a vertical  M3 insert into a bridge in an already designed component, and I'm hoping to avoid redesigning this area (or even worse, having to print in a different orientation). I haven't tried printing the component yet, so I'm hoping it will be OK, but I thought this might give a better change of success.

nophead, 2020-12-15:
Have a look at file:///W:/CHRIS/Projects/RepRaptor/OpenSCADLibs/NopSCADlib/readme.html#Hanging_hole. I use that for printing holes in bridges. I specify the z height and it orients the hole with the direction that Skeinforge does the infill at on that layer. If you use a different slicer you might need to rotate it 90 degrees, or if the fill is along the bridge you might to rotate it 45.

For M2 I can't use a cylinder as it would close up to nothing.

nophead, 2020-12-17:
I have added an alt parameter to polysink, so I will do the same for polycylinder.

martinbudden, 2020-12-17:
I've just added a PR, so you may want to use that.

nophead, 2020-12-17:
Similar to the way I am working on but not quite.

I think it will fail when the layer_height is not representable in floating point. To get around that I overlapped by eps in screw_polysink but then I got issues with slivers at the surface, so I made it exact it the z = 0 layer and mirrored it. So it needs to be different when centre is true or false. It also needs to be rendered when it is a stack to prevent it slowing down OpenCSG.

I do the rotation differently. Basically pass how many alternatives, so 1 would give an alternating interleave every 2 layers and 2 would give three phases every three layers, etc. Although a slow spiral looks nice from the end I think a faster rotation will give a stronger, more uniform hole.

nophead, 2020-12-18:
I pushed my version, which is slightly different. To end the hole neatly at the surfaces it needs to stop at the correct segment because otherwise the next segment makes a scar on the surface due to the overlap between segments. So to put a hole through 10mm you can pass h = 10 and it will actually be eps longer at each end, so avoids z fighting.

I have a twist parameter to specify how many layers it takes to rotate one facet. It also has a slight extra twist so that it does one extra facet over the full length of the hole giving the perfectly round end view even when twist is only 1 and the layers alternate.

martinbudden, 2020-12-18:
I've tried these out, and they do slow things down quite a bit. I have a suggestion, rather than rotate every layer, why not rotate in stacks of `n` layers, where `n` divides the height. So we have something like:

[image]

This would have the advantages of rotating polyholes, without generating nearly as many facets. I imagine it would also make the end calculations easier.

For a hole of depth 10mm and a layer height of 0.25, this would reduce the number of polygons from 40 to (say) 5.
5 seems to be a good number of twists: it is prime and large enough to have a useful effect.

nophead, 2020-12-18:
Not sure it worth doing when it only rotates after 5 layers. Other that giving the impression the hole is round when looking down it, what else does it do when the alternate contact points are so far apart?

I think the alternating case is probably the optimum for printing as the vertices have the maximum area to land on and are less likely to shrink inwards. And when the infill is considered they are tied in above and below giving a much stronger connection for the 3 and 4 sided case.

I don't think I would bother for holes with 5 or more sides as they come out pretty round for me with the bend radius of the filament. Square and triangles would definitely benefit though.

martinbudden, 2020-12-18:
Even if the alternate contact points are far apart, they are still alternate contact points, and so should add strength to the hole, in particular resistance to the bolt being pulled out. But the above is just a suggestion.

I agree that this is not worth using for larger holes. I'm intending to use it for holes of `M3_tap_radius` and smaller. (That's actually a 5 sided hole.)

nophead, 2020-12-18:
We need Stephan of CNCKitchen to do pull tests to see if any of this is worth it!

Someday I might build a test rig like his to experiment with things like this.

I tend to use inserts for everything but I should probably use taps more.

nophead, 2020-12-18:
Thinking about it more: the contact area with the screws is constant. Rotating the rings only distributes it around the screw. If the screw is metal the strength of the bond is determined solely by the plastic, so the distribution around the screw will have little effect. I think the main benefit is that alternating the rings keys them to the infill and that will increase the strength of the joint. I think lumping layers together will dilute that effect proportionately.

martinbudden, 2020-12-18:
Agreed, the contact area with the screw is constant. The gain in strength is from distributing the load around the plastic. Consider a triangular hole: viewed from above there are three contact points with the screw, which translates into 3 vertical lines of stress, all at the same points in the plastic. If the hole rotates, then that load is more evenly distributed around the circle.

I think the effect is probably marginal, however there may be cases when it is important, for example the hole is close to an edge, another hole, or some other stress point. It's all about distributing the stress.

As for the two different schemes, rotating each layer, or rotating a stack of layers, I think differences in strength gains will be second order, however, as you say, you really need to measure this.

What I can tell you for sure, is that having tried the rotating polyholes, they are extremely slow, so slow that I have turned them off. So even if the layer rotation is better, it doesn't matter if it is too slow to be used.

nophead, 2020-12-18:
Slow in preview or when generating STLs?

martinbudden, 2020-12-18:
Both. So the first thing I did was to switch it off in preview. But still generating the STLs was painfully slow. I will turn it back for the final build when I've finished doing the development on my current project. And it's only being used for 24 M3 tap holes, each 8mm deep.

nophead, 2020-12-18:
If all the holes are the same and only 8mm deep I would expect a hit the first you do F5 but that should be cached by the render and it should be quick after that.

nophead, 2020-12-19:
You can stack layers simply by overriding $layer_height before the call. Here is a demo. It does make a big difference to F6.

Preview is only a couple of seconds but it would depend on the complexity of the part it was subtracted from and if it is rendered in the preview. For vitamins I do lots of tricks to speed up the preview. I might subtract a simple cylinder bigger than the hole and union a rendered difference between a cylinder and the twisted hole.  The union is free, and the slow bit is cached and drawn 24 times. Of course that doesn't help F6 because union is expensive.

```
include <NopSCADlib/core.scad>

h = 8;
r = 50;
n = 24;
layers = 4;

difference() {
    cylinder(r = r, h = h);
    
    let($layer_height = layers / 4)
        for(i = [0 : n - 1])
            rotate(360 * i / n)
                translate([r * 0.75, 0])
                    poly_cylinder(r = 1.25, h = h, twist = 1);
}
```

martinbudden, 2020-12-20:
That's a brilliant solution.

I have a wrapper function that solves the F6 problem:

```
function twist(radius) = radius <= M3_tap_radius && !$preview ? 5 : 0;

module boltPolyhole(diameter, length) {
    translate_z(-eps)
        poly_cylinder(r = diameter/2,  h = length + 2*eps, twist = twist(diameter/2));
}
```

(I had my own bolt library before I discovered NopSCADlib, I converted it to NopSCADlib by making it into wrappers around the `screw` modules, so all my bolt and bolthole functions are called via these wrappers. I was intending to go full over to the NopSCADlib `screw` modules, but it turns out that having the wrappers has been useful on more than one occasion.)

I think we can close this issue.

nophead, 2020-12-20:
I am just about to push a small tweak that doesn't add the extra twist when layers is less than twists. So if you pass a large number in twist it does one gradual twist over all the layers like your original. It also make sense when layers is small due to layer_height being large.

One can also pass a large layer height to inhibit twisting but still get the extension by eps. Perhaps poly_cylinder should have always extended by eps but that would be a big change now.

martinbudden, 2020-12-20:
I agree changing `poly_cylinder` now would be too big a change. But you could add a `poly_cylinder_eps` module if you wanted.

closed 2020-12-20 by nophead