nophead/NopSCADlib #114: [Feature Request] Add facility to add text after parts list in readme.html

martinbudden, 2020-12-22:
Quite often it is useful to add text after the parts list, for example to describe possible alternatives for parts, or where to source difficult to obtain parts.

nophead, 2020-12-22:
When we were selling 3D printer kits I had a script that put the parts into a spread sheet where I added URLs to shops and prices to get a costed BOM. It used OpenOffice.

nophead, 2020-12-23:
Perhaps we could split the blurb by looking for the first point number or bullet point as the start of the build instructions. Any lines before that go before the Assembly Instructions heading. Or would you prefer them just after the vitamins list?

nophead, 2020-12-24:
I think a markdown horizontal rule would be a better way of splitting the blurb into procurement instructions and build instructions.

The first rule itself would be omitted but anything before it would be rendered after the vitamin list and anything after it as build instructions.

martinbudden, 2020-12-26:
There are 4 main places to place text:

1. At the very beginning of the readme.
2. In the first section of the readme, but after the picture
3. After the table of contents
4. After the Parts list

It would be useful to be able put text into each of those positions. I think using a horizontal rule as the demarcation works fine.

It would also be useful to specify that the picture is omitted, so that you can put in your own alternative picture.

nophead, 2020-12-26:
Oh good job you clarified because I thought you meant after the vitamins in each assembly step rather than the top level parts list.

nophead, 2020-12-27:
It currently uses assemblies/main_assembled.png for the cover picture. I could look in doc for a file with the same name and use that instead if it exists and also look for .jpg.

nophead, 2020-12-28:
Are you wanting to replace the cover picture with a photo, or another generated image?

I could just omit the picture if it finds a markdown image tag in the first section of the blurb but it sort of confuses the before and after picture description of where the blurb goes. Would it expect one less section when it finds the first section contains an image?

martinbudden, 2020-12-28:
My current thinking is to replace the cover picture with another generated image alongside a photo.

I think the easiest thing is to omit the before and after the picture blurb. If I want blurb after the picture then I can manually insert the picture and have blurb before and after.

So I think we now have the following positions for blurb:

1. At the beginning of the readme. If no image tag in the markdown then the picture is auto-generated, if an image tag is found, then no picture is generated. 
2. After the table of contents
3. After the Parts List

nophead, 2020-12-28:
I pushed a change which I think does what you want. See https://github.com/nophead/NopSCADlib/commit/8d22940506df525ac52ef6f8a00ea03661f3b682

martinbudden, 2020-12-28:
Great, thanks. I'll check it out.

nophead, 2020-12-29:
Note I only look for ***, so if you want a horizontal rule to appear use ---

martinbudden, 2020-12-29:
Yes, I noticed you only look for `***`, I was going to ask you to change it, but on second thoughts it works well the way it is, `---` for a horizontal rule, `***` to separate sections.

So I've tried this, and it works fine.

One minor thing, can you also allow any number of asterisks, ie so that you can have a line of `*************` for the separation, as per markdown syntax.

nophead, 2020-12-29:
Done via https://github.com/nophead/NopSCADlib/commit/4ea08839652e02c42fb033006dde372e01937ac4

martinbudden, 2020-12-29:
Great, thanks. I think this issue can be closed now.

closed 2020-12-29 by nophead