nophead/NopSCADlib #115: [Question] Is there a recommended way to include parts that are not part of the final assembly?

martinbudden, 2020-12-22:
A project may include parts that are not part of the final assembly, for example jigs to drill holes, or to align parts. Also there may be alternative parts, eg the same part with self-taping holes or threaded inserts.

Is there a recommended way to include these types of things in the BOM?

nophead, 2020-12-22:
Yes you can add items on the BOM that are not visible in the assembly view using hidden(). Here is an example
```
//! 1. Strip some brown wire from the cable and use it to link the live pin of the socket to the fuse, using appropriate size heatshrink sleeving at both ends.
//! 1. Add a second brown wire about 100mm long to the output of the fuse and add heatshrink sleeving.
//! 1. Connect 3 core mains lead to the earth and neutral on the IEC and leave about 100mm of the brown live wire free.
//! 1. Feed the cable in through the IEC hole and out through the hole in the back of the box with the two free brown wires comming out of the switch hole.
//! 1. Place the IEC inlet into the box and secure with two M3 x 10mm countersunk cap screws.
//! 1. Add heatshring sleeving to the two brown wire and solder then to the switch.
//! 1. Press the rocker switch into the panel in the orientation shown.
module mains_inlet_assembly() pose([215, 0, 215])
assembly("mains_inlet") {

    hidden() {
        vitamin(": Mains lead, 3 core, 6A 24/04 0.75mm&sup;2, 300mm");

        for(i = [1 : 3])
            tubing(HSHRNK64);

        for(i = [1 : 4])
            tubing(HSHRNK32);

        vitamin(": Fuse, 20mm 5A T");
    }

    mains_inlet_box_assembly();

    translate([iec_x, iec_z])
        hflip()
            iec_assembly(mains_inlet, 0);

    translate([switch_x, switch_z])
        hflip()
            rocker(mains_switch);
}
```
[image]

nophead, 2020-12-22:
```
Qty | Description
-- | --
1 | Fuse, 20mm 5A T - not shown
4 | Heatshrink sleeving ID 3.2mm x 15mm - not shown
3 | Heatshrink sleeving ID 6.4mm x 15mm - not shown
1 | IEC fused inlet old
1 | Mains lead, 3 core, 6A 24/04 0.75mm⊃2, 300mm - not shown
1 | Rocker Switch PRASA1-16F-BB0BW
2 | Screw M3 cs cap x 10mm
```

martinbudden, 2020-12-22:
It seems that `hidden` only works for vitamins, not printed parts. Is that right?

nophead, 2020-12-22:
I think it will hide anything. The only thing it won't do is add "not shown" to the description because printed parts don't have a description.

It is a bit bodgy because it hides them by making them a million times smaller.

martinbudden, 2020-12-22:
Thanks. Yes, that was the bit that confused me - not adding "not shown" to the description.

I'll close this.

closed 2020-12-22 by martinbudden