nophead/NopSCADlib #130: ValueError: invalid literal for int() with base 10 error in tests.py

martinbudden, 2021-01-12:
I'm sometimes getting this error

```
 File "../nopscadlib/scripts/tests.py", line 59, in compare_images
    pixels = int(f.read().strip())
ValueError: invalid literal for int() with base 10: '1.06957e+006'
```
Seems to be an issue with how python deals with literals, floats and ints, see https://stackoverflow.com/questions/1841565/valueerror-invalid-literal-for-int-with-base-10

I think you need to use `pixels = int(float(f.read().strip()))`

nophead, 2021-01-12:
Yes if the number of pixels is not an integer it will fail. Not sure how an image can differ by a shade over a micropixel though!

I will fix is as your suggestion.

nophead, 2021-01-12:
Should be fixed via https://github.com/nophead/NopSCADlib/commit/ac6e8e040aa6a0789013355fa00b7b0f99253cad

Not sure how to test it. How did you manage to get a fraction of a pixel difference?

martinbudden, 2021-01-12:
1.06957e+006 is 1,069,570, so it's not a fractional difference, just a large number that so that exponent notation was used.

Not sure why I am getting such large numbers, the things I am comparing are quite complex - PCBs with a lot of components in an assembly.

The fix seems to work, I'll do a full build later and confirm.

nophead, 2021-01-12:
Yes that makes more sense. A scene moved a few pixels could end up with millions of pixels different.

martinbudden, 2021-01-13:
This seems to be working fine now. I think the issue can be closed.

nophead, 2021-01-13:
OK I will close it. As the owner of the issue can you close it yourself instead of saying it can be closed?

closed 2021-01-13 by nophead

martinbudden, 2021-01-20:
Generally I think it is good not to close your own issues. That way you ensure that at least two people agree that the issue has been fixed.