nophead/NopSCADlib #134: Countersunk screws do not lie entirely flush

martinbudden, 2021-01-14:
I'm finding I have to offset my countersunk screw holes by `layer_height` to get the screws to lie completely flush (fortunately this is possible because the countersink cutout is reflected in the xy plane).

I think the counterbore associated with the countersink is off by one layer. This may be a rounding issue - I tend to print with a layer height of either 0.4 or 0.5. Certainly I am seeing this with a 0.4 layer height, I haven't checked for 0.5. I suspect that a calculation that is correct for a layer height of 0.25 may be out by one for a layer for a layer height of 0.4.

nophead, 2021-01-14:
Have you set £layer_height before including NopSCADlib? On my phone so don't know how to type dollar.

martinbudden, 2021-01-14:
Yes. I have a file `global_defs.scad` which sets `$nozzle`, `$layer_height`, `$extrusion_width` and the `$ppN_colour`s which I include in every `.scad` file before I include `<NopSCADlib/core.scad`.

nophead, 2021-01-14:
It works by sampling the function that describes the shape by the layer height, so perhaps your screws don't match that model that I derived from online formulae. It is hard for me to test anything as I don't have any real screws or a 3D printer here.

martinbudden, 2021-01-14:
OK, let's leave it outstanding for the moment. I might look into it a bit later on, but I don't want to go down this rabbit hole just at the moment.

nophead, 2021-01-14:
I think the magic number is here: https://github.com/nophead/NopSCADlib/blob/master/vitamins/screw.scad#L274

The thickness of the non-conical top of the head is supposed to be 1/5 of the screw radius. If it is thicker than that it would sit proud. You could try making 1/4 and see if it helps.

Of course any over extrusion would have the same effect. The cone should touch the filament's rounded edges tangentially, so I would expect it could be snugged up a little deeper, which is why I didn't add any extra clearance.

martinbudden, 2021-01-14:
I've just checked some real screws. For M5 countersunk the head diameter is 9.95mm and the counterbore is less than 1mm. For M3 screws in from the same batch the head diameter is 6mm, but the counterbore seems a tad more than 0.6mm, but I don't have the capability to measure these small values precisely. So I think the divisor of 5 is correct.

I'm definitely not over extruding. For prototyping I print at 0.4 or 0.5mm layer height and and 0.8 or 1.0mm extrusion width which definitely pushes the limits of the standard hotend and results in under extrusion.

I'll check the calculation to see if there is a rounding issue.

nophead, 2021-01-14:
Yes but its is divisor of radius, not diameter, so for M5 it would be only 0.5mm and M3 would be 0.3mm, so perhaps it needs doubling. That would mean the screw model would also change,

martinbudden, 2021-01-14:
My measurements show that the counterbore is 1/10 of the head diameter, that is 1/5 of the radius. So the radius divisor of 5 that you use is correct.

nophead, 2021-01-14:
Yes but I am dividing the screw radius by five and that is half the head radius.

nophead, 2021-01-14:
Actually I found an M4 cs cap screw and I have a cheap digital caliper but it isn't easy to measure. I think I need a surface gauge but I haven't got anything more than very basic tools here.

By placing inside an M8 by 0.5mm brass washer from a clock it looks to be higher than the flat section of the head, so I can believe it is about 0.4mm, which makes the code correct but your screws seem to be thicker.

martinbudden, 2021-01-17:
It's possible that my screws are thicker, but I think it is more likely that my measurements are inaccurate - I don't have the tools to make these kind of measurements accurately.

I think we can assume the code calculating the counterbore is correct. That leaves the conversion into a number of layers.

I'm definitely seeing my M5 countersunk screws stand proud. An M5 cs screw should have a counterbore of 0.25mm, so if the layer_height is 0.4mm do you round that up to 1 layer, or do you round it down to 0 layers?

nophead, 2021-01-17:
I don't do any rounding. I sample the infinite cone at each layer and compensate for the fact the contact is on the rounded edge of the layer. I then limit it to lie between the head size and the screw clearance size. So the counterbore just comes from the limit and the fact the cone is offset downwards, so it hits that limit before it gets to the surface.

The radius is 2.5mm, so the counterbore should be 0.5mm, not 0.25mm.  I.e. 1/5 of the screw radius, or 1/10 of the screw diameter, or 1/5 of the head diameter.

So the top layer at least should clear the head. If you are using 0.4 layers I think the first two should clear the head giving a 0.8 counterbore but the screw will site higher than that because it sits on the cone.

nophead, 2021-01-17:
This is what a simulation looks like with layer_height = 0.4 and extrusion_width = 0.8.

[image]

The second layer is fractionally smaller than the counterbore because it is predicted to land on the cone but in fact the cone stops short of it, so it could be slightly smaller.

[image]

This is before conversion to polycircles. These are smooth circles of the same radius.

This is what the polycircle STL looks like.

[image]

If you place the screw head in upside down it should easily bury the flat bit.

martinbudden, 2021-01-17:
I've just printed this to try it out. You are correct, if I place the screw head in upside down, then the flat bit is easily buried.

However if I put it in the right way around, the screw stands ever so slightly proud, maybe 0.1 or 0.2mm. For the thing I am using this for, this is important, because I am bolting a plate on top of the screw, so it absolutely cannot protrude.

I think the reason is, as you said, because the screw is resting on the cone.

nophead, 2021-01-17:
You are using screw_polysink() and not screw_countersink()? Also are these holes at the top, or underneath?

The only other reason I can think is the screw doesn't have a accurate 90 degree cone. Metric screws should have but imperial are a different angle. And there is quite a lot of tolerance on screw dimensions, so perhaps the hole should a bit lower to account for that. It is better to be below than it is to be proud.

As a workaround perhaps you can lower it by 0.2mm. As it is double ended I think that will just make it deeper, even with 0.4mm layers as each ring will get a bit bigger, so it should drop by 0.2mm.

When I get home in March I can do some experiments.

martinbudden, 2021-01-17:
Yes, I am using screw_polysink. I'm using metric screws and the hole is on the top of the part.
 
I have bought screws from different suppliers on different occasions and have noticed a difference in quality of the finish, so presumably there is variation in dimensions. I agree it would be good to allow for some variation in dimensions.

My workaround is just use `translate_z(-layer_height)` on the desired screw hole. This works fine, since the polysink is symmetric about z=0, and so I am "stealing" a layer from the "reflection" of the polysink. But it would be nice not to have to do this.

nophead, 2021-01-17:
Yes that I agree that is a workaround. I think somehow we need to work out the maximum dimensional error and have that as the default. It is  combination of the the angular deviation from the 90 degrees and the diameter variation of the head and the possibly the tolerance of the top thickness.

nophead, 2021-03-22:
I printed the test STL made by tests/screws.scad and my screws fit perfectly flush when tightened by a nut. I was surprised how accurate it was.

[image]

Hard to photo but I can slide a straight edge over the top without it snagging on the heads.

When fitted to the underside countersinks some screws were a bit proud. Presumably the extrusions sag a bit when overhanging or perhaps slope inwards a bit. 

So I added a `sink` parameter that lowers the cone and printed the test again with it set to 0.2mm. The screws then fit with their heads sitting below the surface close to 0.2mm, about 0.15mm to 0.25mm.

[image]

The test now has a customiser slider to experiment with. I am not sure if your screws are non-standard or you print process is different but you should be able to compensate with this parameter. Note it doesn't need to be a multiple of layer height. I tested it at 0.2mm with a 0.25mm layer print.

martinbudden, 2021-03-31:
> When fitted to the underside countersinks some screws were a bit proud.

I hadn't thought about that. Virtually all of the countersinks I have printed have been on the underside. I've retried M5 and M3 countersinks on the top side and they are indeed flush.

One other difference in my prints is that I tend to print everything with a 0.5mm layer height. The countersinks work fine at this layer height as well.

The `sink` parameter handles the case for printing on the underside, so I think this can be closed now.

nophead, 2021-03-31:
While answering a question about print accuracy in another forum I measured the outside dimensions of the test piece. I was amazed that my machine, which hasn't had any calibration for years, prints to a tolerance of 0.1% with this glow in the dark ABS that happened to be loaded.

[image]
[image]

Where the screw sits will be a function of X Y and Z calibration, particularly in the underside because the first layer height will be directly significant.  My part measures 10.08mm high and I only need 0.1mm for `sink` to make them sit flush in the bottom. But my first layer is probably 0.08mm too high, so 0.2mm is probably a more reasonable value. Perhaps more if you squish your bottom layer.

closed 2021-03-31 by nophead

martinbudden, 2021-03-31:
> I was amazed that my machine, which hasn't had any calibration for years, prints to a tolerance of 0.1%

Well, there's no reason for a well built machine to go out of calibration just because of the passage of time. As long as you don't actually knock it, or subject it to extremes of temperature it should stay calibrated. But on the other hand, I can understand the amazement. I find 3D printing is full of quite moments of amazement. I just recently designed a fan duct for the 30x30 blower, using just the dimensions from NopSCADlib, and was amazed that it fit perfectly first time.

Although my machine is properly calibrated, I can't guarantee that someone using one of my STL files has a properly calibrated printer, and in particular people do vary their bed offset with different filaments, so the ability to add some tolerance into the STL part is very useful.

nophead, 2021-03-31:
I have my slicer XY scale set to 1.0057, which is a value I measured more than 10 years ago on a machine built with ball screws, so it has precise motion to 6um resolution.  I can't remember what plastic that was was derived from, I think it might have been PLA and this was done with ABS on a machine with belts. 

I don't calibrate the axes, I just use 80 steps per mm from the pulley calculation. There must be some difference depending on how much tension is on the belts. I plan to investigate that on my hbot machine that only has one belt, so at least X & Y will be exactly the same.

I have got used to everything I print fitting exactly and matching my milled parts. The only time they didn't was when I accidentally applied the scale factor twice, once in the slicer and once in my own host that I use to run my older machines.