nophead/NopSCADlib #138: Use of absolute path in rounded_right_triangle.scad

damienmg, 2021-01-23:
In https://github.com/nophead/NopSCADlib/blob/e41366e20c83875665817b8a9c25cf7d52da7429/utils/rounded_right_triangle.scad#L24, the path used is an absolute path which breaks imediately when the lib is not directly in /NopSCADlib 

I noticed this since OpenSCAD was complaining about not being able to open rounded_rectangle.scad.

nophead, 2021-01-23:
Fixed via https://github.com/nophead/NopSCADlib/commit/c9e99420709b1ac0da0026970bc3934f2bdf8559.

It wasn't technically an absolute path and should be found because NopSCADlib should be in your OPENSCADPATH, which is why I hadn't noticed it. It does cause a performance issue though because OpenSCAD sees different paths to the same module which causes duplicate module cache entries, which is why I always use ../ within the library and NopSCADlib/ outside of it.

nophead, 2021-01-23:
Thanks for pointing it out.

closed 2021-01-23 by nophead