nophead/NopSCADlib #148: Location of temporary build files

martinbudden, 2021-02-08:
NopSCADlib creates a number of different temporary files during the build (eg `tmp.png`, `png.scad` etc). These are created in the root directory of the project, which creates at least two problems

1. If you interrupt the build, the temporary files need to be cleaned up.
2. You have to add the temporary files to the `.gitignore` of the project.

I suggest the temporary files are created in a `tmp` directory, so the root directory isn't cluttered by them. Specifically, for projects without variants, the directory `tmp` would be used, for projects with variants the directory `variant/tmp` would be used.

nophead, 2021-02-08:
I don't think either of those are true because if you interrupt a build the next build will remove it.

They should probably be in the system tmp folder with unique filenames but then you do get a problem with them being left being.

martinbudden, 2021-02-08:
I frequently have temporary files lying around. If I interrupt a build it's often because I realise I have to change something, so it might be some time before I do another build.

Also because builds take so long, I often do editing with a build going, so I get temporary files popping up in the difference pane of vscode, which is annoying, so yes they do need to be added to `.gitignore`

Another advantage of this, is by putting the variant files in `variant/tmp` you can simultaneously build two variants, each in a separate command window - if you currently try and do this you get problems because the temporary files clash.

nophead, 2021-02-08:
Why does putting it in a tmp folder mitigate the need for .gitignore? Don't you still need to add at least the folder?

I can use `tempfile.TemporaryDirectory()` which will make a unique named directory and delete its contents on exit but won't you get that popping up in vscode if it has a random name?

Possibly I can just use a Python destructor to ensure the file is removed when the build terminates.

What about openscad.log and openscad.echo that are useful after the build has finished?

martinbudden, 2021-02-08:
Yes, I'd have to and the tmp file to .gitignore. But that's a simple operation, and is resilient to new types of temporary files created in future. In the current setup I had to find out all the types of tmp files and add them individually to .gitignore.

Yes, `tempfile.TemporaryDirectory()` would cause the same issue with vscode. So it would be better to use  tmp as the directory, which is then deleted when the build finishes.

For openscad.*, they can stay in root, since they are not really temporary files.. For variants, though, I'd like them to go in the variant directory.

nophead, 2021-02-08:
You can't build two variants at the same time because the target is set by writing to target.scad.

martinbudden, 2021-02-08:
Oh yes, I'd forgotten that.

martinbudden, 2021-02-08:
But it would be possible to run the tests and a build of a variant simultaneously.

nophead, 2021-02-08:
Only if the test didn't include target.scad. I.e. if it was invariant with respect to the build target.

martinbudden, 2021-02-08:
Agreed. But tests should be designed like that anyway - they should explicitly test the variants, rather than do it implicitly - ie they should include `config_variant.scad` rather than `target.scad`.

nophead, 2021-02-08:
Actually I just debugged a problem by using the fact the temporary files are left around after an error. Having them deleted would make it harder to debug errors.

nophead, 2021-02-08:
So does just moving them to a target/tmp folder but only deleting them if the command succeeds work for you? 

I could make the names reflect what is was making. E.g. tmp/main_assembly.png and of tmp.png?

martinbudden, 2021-02-08:
Yes, that would be great.

Having the names reflect what it is making might turn out useful in debugging.

nophead, 2021-02-09:
Should be implemented by

https://github.com/nophead/NopSCADlib/commit/fc44b43638914a991117477425cbb2f631e9dcae

I have to say all these changes are stopping me using my own library and the build system is starting to get hard to maintain without bugs although a lot of that is the big change I made to speed up assembly view pngs, which is actually a benefit to me.

martinbudden, 2021-02-09:
Yes, that works fine. Thanks for implementing this.

I'm not quite sure what you mean when you say the changes are stopping you using your own library. Do you mean they have broken some of your code, or do you mean you are spending time doing these changes, rather than using your library?

nophead, 2021-02-09:
I mean the constant stream of requests and PRs all take time to process and I keep finding bugs in the build system and have to fix them due to it getting ever more complicated. I like to keep things simple.

martinbudden, 2021-02-09:
With regard to the stream of requests - you can always say "no, I don't want to do that", or "I'll do that later". But I don't have any further requests for changes to the build system. I very much appreciate that you have made the changes you've made, and hope they may be useful to other people as well. I agree that it's good to keep things simple.

With regard to PRs, I've been going through my projects and making a PR for anything that I thought might be generally useful to others. I've only got two things left, one is an Extruder vitamin, and the other is a parameterised CoreXY belt utility. I don't plan on submitting either for a while though, the Extruder because I want to improve the drawing and the CoreXY because I want to spend some more time checking the various permutations and that the helper functions are correct and sufficient.

nophead, 2021-02-09:
Yes the additions are appreciated. I used your square blower because I was having trouble finding space for the ones I have and then when you added the smaller one I switched to that!

closed 2021-02-11 by nophead