nophead/NopSCADlib #151: how to use pose? and zoom?

SmoothieAq, 2021-02-16:
I find it hard to figure out which values to use for pose() - do you have some hints to do that without the need to do make_all to see the result?
I can not figure out zoom, so I only show some part of an assembly. I would like to zoom in where a small part is fitted onto a much larger part. Is there a way to do that?

nophead, 2021-02-16:
If you view the assembly in the GUI and use that to pan, and rotate you can use the translate and rotate values in the status bar for `pose()` to give the same view. You can use Edit / Copy viewport translation and Edit / Copy viewport rotation to copy those values and paste them into your code.

There isn't a way to zoom though because when the assembly views are created `--viewall` is passed on the command line to make the whole assembly fit. 

If the assembly picture is small you can force a big one by passing `big = true` to `assembly()`

Perhaps you could crop the large part of the assembly when `is_undef($pose) || !is_undef($posed)` to cause that to be zoomed into. `clip()` might be useful for that.

SmoothieAq, 2021-02-17:
Thank you!

I now understand the pose arguments, and the edit/copy hint is nice. I do however still find it a bit difficult to do because of the later --autocenter. Do you know if it is possible to do the equivalent of autocenter in the viewer?

The clipping it a nice hint! I take it, that I should clip when `!is_undef($pose) && is_undef($posed)`?
With my model clip() never finishes, it just eats memory. However, I successfully clip with difference() and some boxes.

I know I am now asking for generic OpenSCAD help, but I have a problem when posing a clipped model containing extruded elements. And many (/most) vitamins in NopSCADlib are extruded.

Here is an example:
```
difference() {
	union() {
		extrusion(E3030, length = 100, center = false);
		translate([60, -15, 0]) cube([30, 30, 100]);
	}
	translate([-20,-20,-10]) cube([120,100,60]);
}
```
[image]

Do you now of a way to avoid this strange clipping of the extrusion?

Anyway, consider me helped already, and just close this whenever you want! If you have the time you might want to add some of this to the documentation :-)

nophead, 2021-02-17:
The GUI viewall button is the same as the --viewall option used to make the images. It points the camera and sets the zoom.

Yes if you clip when !is_undef($pose) && is_undef($posed) is true then it will only be done in the assembly view of that assembly and not when the assembly is part of another view and not in the GUI.

clip() does an intersection with a big cube and renders the difference, so it should always finish but if the object is complex it might be slow and colour will be lost.

None of the vitamins are designed to be cut, so they don't set the convexity parameter of linear_extrude. To display a convoluted shape like an extrusion it needs to be set if subsequent operations are done, or the result has to be rendered, like clip does.

Is it just an extrusion you want to clip() ? If so perhaps the convexity parameter should be set there, or perhaps be a parameter.

nophead, 2021-02-17:
The reason this stuff isn't documented is I haven't anticipated ever needing to zoom in an assembly view. The biggest thing I have made is a 3D printer and I haven't needed to do that but I can see once you start using extrusions you can end up with very large assemblies. Maybe we need an option to stop --viewall and add the distance parameter to pose(). That is probably more practical than clipping.

SmoothieAq, 2021-02-17:
Yes, a convexity parameter fixes it, thank your!
I think it would be nice if the potential "long" extruded vitamins all took a convexity parameter to their rendering module.
It would also bee very nice with a distance parameter to pose() :-)

nophead, 2021-02-18:
Yes I think perhaps I can make adding the d parameter to pose() suppress --viewall when it is set.

nophead, 2021-02-20:
I have implemented in in this commit: https://github.com/nophead/NopSCADlib/commit/7c2df8d36d214a8bf8ebdcd39da293209bc175ab

Instructions for using pose() have been added to BOM.

SmoothieAq, 2021-02-21:
It works great! And easy to use! Takes out the unpredictability of big/small and viewall :-)
Thank your!

closed 2021-02-21 by SmoothieAq