nophead/NopSCADlib #153: feature request: make clean

SmoothieAq, 2021-02-21:
To remove all that make_all makes.

nophead, 2021-02-22:
What is the use case you have in mind?

I can easily remove all the directories, like bom, stls, deps, etc, plus a few files like readme.* but that will also remove anything placed there by the user. For example I have sliced gcode in my stl and dxf directories.

I could remove just the file types expected, e.g. .STLs from stls and then only remove the directory if it was empty. That would leave sliced files.

More difficult would be removing just the stls, etc. the model currently creates.

SmoothieAq, 2021-02-22:
Ok, personally I prefer to clearly separate what is generated and what is "source".

It is a bit of another case, but I would actually prefer if make_all made all its outputs somewhere else. Then I could have one git repository with my source, and another with nice accessible readme etc. And my commits could then be of either source or of output. As it is now, my commits mix source and some version of the make_all output, that may or may not be up to date with the source. I would think this is a pretty common way to think about software and "compilation" and "generations".
If you by into that, you would have sliced gcode in some other place, at this is a case where the stl's are source, and gcode are the output (from the slicer).

Anyway, in my experience, when I upgrade one of my libraries (for example NopSCADlib), I need to delete stuff to make sure all is generated again. Perhaps it is enough to clean some dependency files? Well, this is my most important use case. As an example, I thought I met a bug in your nice new pose(), but when I deleted everything and ran make_all again, all was good.

nophead, 2021-02-22:
Well where the output goes is a another topic entirely. I like everything under the same tree and only check in a project after I have done a make_all.  Although STLs are not source I put them on Github, so people can make the project without building it from source.  If you don't want to add the STLs, etc, you can just add the directories to .gitignore.

Source is separated from generated files because they are in the scad folder. The only generated stuff that isn't in its own folder is the readme because it has to be in the top level for github.

Deleting the dependencies does not force everything to be generated because a single source file can be responsible for many STLs and DXFs. The missing dependency will only force the first STL to be generated. Similarly it can be responsible for many assembly views. Deleting times.txt /s probably forces everything to be built.

In general the dependency checking will update everything for source file changes. It won't handle things like OpenSCAD changing, or if the build system changes what is generated. Pose is the only case I can think of but in the case of pose the results should only change when you add a d parameter, so that source change should be enough to regenerate the views affected when you run make_all. The fact that d has been set is communicated to views.py via bom.json, so bom.py needs to be run before views.py, but make_all.py does that. 

I am minded to do my second option. Remove all the stls and pngs, etc from the stls directory and then delete the directory if it is empty and the same for the others.

martinbudden, 2021-02-27:
There is a way of achieving this with the current build system. If you make give your project a configuration (see https://github.com/nophead/NopSCADlib/blob/master/docs/usage.md#multiple-configurations) then you can just delete the configuration directory to get a clean build.

It may seem a bit odd to use configurations even if the project has only one configuration, but it does work and results in a cleaner directory structure. Indeed, I've started using configurations for any project of any reasonable size, even if there is only one configuration.

SmoothieAq, 2021-02-27:
Thanks, @martinbudden! That is exactly what I was looking for, just didn't realize I could use configurations for that.

closed 2021-02-27 by SmoothieAq

nophead, 2021-02-27:
Yes that would bundle all the build artefacts including the readme one level lower than the source. So you could publish a project somebody could build without source. The only problem is if you did want to publish with source then the readme would be in the wrong place for Github.

For multiple config builds I should probably generate a top level readme linking to the others. That might upset @SmoothieAq again though.

I published NopSCADlib so that I could put projects in Github but, other than the simple example, I haven't actually done that yet. The problem is most of them are too dangerous for general consumption or I haven't kept sufficient details of the electronics.

martinbudden, 2021-02-27:
Actually, for multiple config projects, the it works fine as it is - the top level readme can be manually written with an overview of the project, and can link to the config readmes which give the build instructions, bom etc.

nophead, 2021-02-27:
Yes that would work fine. You would just have to remember to add any new variants manually but not a significant burden.