nophead/NopSCADlib #154: feature request: change log

SmoothieAq, 2021-02-21:
I have been using NopSCADlib on/off for a couple of years, and love it!!!

However, when I infrequently upgrade to the current version, I have a bit of detective work to fix what no longer works (lastly it was rails). The changes always seems to the better, but it would be very nice with an ongoing change log that briefly listed braking changes and nontrivial new features. I find the git commit-log hard to read.

nophead, 2021-03-03:
OK I have added semantic version tags that mark breaking changes, backwards compatible additions and fixes that don't change the interface. From these tags I automatically generate a CHANGELOG.md that shows all the commits but groups them under version headings. So you can review all the major version changes from the date you last downloaded the library to see what breaking changes you will encounter updating.

Each version heading links to the tag and you can download the version as a zip file from there. You can also click on the three dots to see all the code changes from the last version that has been tagged.

Commits link the the commit, issues link to issues, etc.

Let me know what you think of the format.

SmoothieAq, 2021-03-04:
nice! easy to spot breaking stuff, and fairly easy to browse for new stuff.

closed 2021-03-04 by SmoothieAq

martinbudden, 2021-03-04:
This looks really good. Is there a convention that should be followed when submitting PRs?

BTW, there is a typo on line 49 of `changelog.py`, "wod" is used instead of "word".

nophead, 2021-03-04:
My research showed the normal way to do this is with structured commit names that indicate change, addition or fix. That is used to automatically bump the version tag. As I have 800 legacy commits and git makes it very difficult to change commit names once pushed I decided to do it the other way around. I manually added the version tags in the git GUI, which wasn't too hard. I lumped together changes of the same type, so not as many tags as commits.

Going forwards it would help if commit names start with Changed, Added or Fixed. Possibly the new tags could then be added automatically although it isn't too onerous to do it manually.