nophead/NopSCADlib #156: Not all directories created when building variant from scratch

martinbudden, 2021-02-26:
More specifically, neither the `variant/dxfs/routed/` directory nor the `variant/stls/printed/` directories are created when doing `make_all.py variant`.

I think this a relatively new bug. The only thing placed in these directories is the `times.txt` file, I think until fairly recently the `times.txt` files were placed in the `variant/stls/` and the `variant/dxfs/` directories.

nophead, 2021-02-26:
It was a recent bug but it should be fixed by https://github.com/nophead/NopSCADlib/commit/e0b89359aabb05a5c35f85a4036a1d055a02e4ad

nophead, 2021-02-26:
See https://github.com/nophead/NopSCADlib/issues/152

martinbudden, 2021-02-27:
I've checked this and it has indeed been fixed. This can be closed now.

closed 2021-02-27 by nophead