nophead/NopSCADlib #159: writing-mode: vertical-rl not honoured in readme.md Parts List table

martinbudden, 2021-03-05:
The Parts List table uses `writing-mode: vertical-rl` for the table column headers. While this is honoured in the the generated `readme.html`, it is not honoured by github markdown, and so does not work for `readme.md` files uploaded to github.

See https://github.com/nophead/NopSCADlib/blob/master/examples/MainsBreakOutBox/readme.md for an example.

This could be solved by generating the Parts List table as an HTML table, rather than a Markdown table.

SmoothieAq, 2021-03-05:
There are lots of variations of markdown out there.
I would prefer that we keep it to the basic facilities. I write OpenSCAD in IntelliJ with the OpenSCAD plugin (all free, and although the plugin is quite buggy, this works really well for me). Anyway, the build-in markdown editor in IntelliJ only support the core markdown, and does not support HTML. A good source for what is https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet . If you keep it to the facilities in that cheatsheet, you are pretty sure it will format well everywhere.
If you think nice formatting is more important than keeping to the core markdown, then I agree with @martinbudden, that you might as well embed some HTML.

martinbudden, 2021-03-05:
@SmoothieAq , the `readme.md` file is a generated file, not something the user edits, so it would not affect your ability to use IntelliJ for editing your project.

nophead, 2021-03-05:
Github strips out attributes from HTML style tags so you can't do anything like colours, etc. For example that table looks like this on GitHub:
```

<table>
<thead>
 <tr>
 <th align="right"><span>Base</span></th>
 <th align="right"><span>Feet</span></th>
 <th align="right"><span>Mains In</span></th>
 <th align="right"><span>Main</span></th>
 <th align="right"><span>TOTALS</span></th>
 <th align="left"></th>
 </tr>
 </thead>
```

This is what Markdown produces:

```
<table>
<thead>
<tr>
<th align="right"><span style="writing-mode: vertical-rl; text-orientation: mixed;">Base</span></th>
<th align="right"><span style="writing-mode: vertical-rl; text-orientation: mixed;">Feet</span></th>
<th align="right"><span style="writing-mode: vertical-rl; text-orientation: mixed;">Mains&nbsp;In</span></th>
<th align="right"><span style="writing-mode: vertical-rl; text-orientation: mixed;">Main</span></th>
<th align="right"><span style="writing-mode: vertical-rl; text-orientation: mixed;">TOTALS</span></th>
<th align="left"></th>
</tr>
</thead>
```

So the only way to get it to work in Github would be to generate images.

martinbudden, 2021-03-05:
That's a pity.

On a related subject, why do you use non-breaking spaces in the table headers? Using normal spaces is certainly better for non-vertical text in the header, and even for vertical text I think it might be better.

nophead, 2021-03-05:
Because I want the columns to be as narrow as possible so that they fit on a printer page. Even then I had to lump assemblies together using the ngb parameter.

[image]

Looks like Github also strips those so when it is in horizontal mode they can wrap leading to narrower columns.

martinbudden, 2021-03-06:
It looks like, given the striping done by Github, there is nothing that can be done to improve the display of `readme.md` in Github, so I think this can be closed.

(I'd forgotten about the `ngb` parameter - using that will help a lot.)

closed 2021-03-06 by nophead