nophead/NopSCADlib #166: 2021.01

SmoothieAq, 2021-03-10:
I am going to try to make a PR.
Would you consider lifting the OpenSCAD requirement to 2021.01?
I find, that the new ability to make functions in a local let-scope is really nice :-)

```
function foo(x) = let(
        z = 7,
        bar = function(y) y+x+z
    ) bar(17);
```

nophead, 2021-03-10:
I am not averse to updating requirements to a later release but not sure it is worth it just for syntactic sugar. You could define `bar()` as another function, prefix it with _ to avoid namespace pollution and not document it with //! to hide it from the user interface. On the other hand if you were passing function literals to customise something, e.g. the comparison operation for a sort that can't be done any other way.

For me updating OpenSCAD is trivial and I run a recent snapshot to get vbo rendering as it makes a huge difference. I don't know how many people just use the version packaged for their Linux distribution, which tend to lag behind.

SmoothieAq, 2021-03-10:
yup. I make the PR, and then you can have a look. As you write, it is pretty trivial to use global _functions instead.
Guess I'll better try a snapshot :-)

closed 2021-03-10 by SmoothieAq