nophead/NopSCADlib #167: PR

SmoothieAq, 2021-03-11:
Sorry to make an issue for this:
I have tried to submit a PR, but got an unable to access error.
I have never tried to do a PR before, ever, so I am probably doing something very basic wrong - any chance you know about a tutorial that will work for submitting a PR to NopSCADlib.

nophead, 2021-03-11:
There is nothing different about NopSCADlib compared to any other Github project.

You make a new branch in your fork of the project and GitHub will add a button to it that will generate a PR. You shouldn't need any special permissions because I have to merge it before it affects anything.

SmoothieAq, 2021-03-11:
ok, I think I made it now. Not sure why I got errors in first try. Sorry for the trouble.

closed 2021-03-11 by SmoothieAq