nophead/NopSCADlib #169: MGN7C and MGN9C carriage dimensions possibly slightly incorrect

martinbudden, 2021-03-31:
I was looking at adding MGN7H and MGN9H carriages and noticed the dimensions of the MGN7C and MGN9C carriages were slightly out. Currently we have:

```
//                  L     L1    W   H   H1   C   B
MGN7C_carriage  = [ 23,   14.3, 17,  8, 1.5, 8,  12, M2_cap_screw, MGN7 ];
MGN9C_carriage  = [ 29.7, 20.8, 20, 10, 2,  10,  15, M3_cap_screw, MGN9 ];
```
whereas I have checked several sources and they say
```
MGN7C_carriage  = [ 22.5, 13.5, 17,  8, 1.5, 8,  12, M2_cap_screw, MGN7 ];
MGN9C_carriage  = [ 28.9, 18.9, 20, 10, 2,  10,  15, M3_cap_screw, MGN9 ];
```

nophead, 2021-03-31:
Looks like I measured the ones I have as they match those dimensions, not the datasheet values that my order links to.  Either that or the datasheet has changed since I ordered them 4.5 years ago.

I am not surprised the thickness of the plastic end caps varies between manufacturers but I am surprised the metal block length varies.

Making the total length smaller would increase the maximum travel but that would allow the wiper to go past the end of the rail with my parts. Making the block length smaller would make the plastic parts that mate with it a little undersized.

martinbudden, 2021-03-31:
In my use cases (so far) my mating parts have deliberately exceeded the length of carriage (to support sensorless homing), which is why I had not noticed this before.

I think generally speaking, the length dimensions are non-critical, whereas it is essential that the width, height and hole spacing dimensions are correct.

I'll leave it to you as to whether you want to make any adjustments to the lengths, in any case, I think you can close this.

nophead, 2021-03-31:
Technically all dimensions like these should have a tolerance and then you could use the min, max or nominal as appropriate. So far I have only sort of done that for microswitch trigger positions, where I had to ensure the axis has enough travel to trip the switch in the worst case. I don't really want to get that complicated in general. In this case being the maximum value ensures designs will work but may be not quite optimum.

So I think the values can stay as they are until someone finds a reason it would break their design. We would then need two versions of the carriages I think.

closed 2021-03-31 by nophead