nophead/NopSCADlib #174: support on parts

SmoothieAq, 2021-04-13:
Would love some advice:

I have some parts, where it is beneficial to model some select supports in OpenScad. These supports are then meant to be cut away with a knife after print.
But I would like the parts to show in the assemblies without the supports.

How best to do that? I love the speedup of cached stl's with `module xx_stl stl("xxx") model();` but then it seems I will need two different stl's?

Thanks!

nophead, 2021-04-13:
You can guard the supports with `show_supports()`, which shows them in the exploded view but not the assembled view. So you can add an assembly instruction to remove the support and you see it both in place and removed.

However, as you point out, that is not compatible with using the cached STLs, as you need two. To do that add a parameter to the STL to show the supports and default it to `false`. That will remove it from the assembly views and create an STL without the support. Then you add a file to `platters` that invokes the stl maker with the support parameter set to `true`. Give it a name like `thing_supported_stl()`. 

A soon as you have any platters you get an `stls/printed` folder that only contains the STLs that need printing. It will include `thing_supported.stl` but but not `thing.stl`.

I also use this technique if I want to rotate an object when printing it as well as when I want to group objects for printing, which is the primary use. The only downside for support is you then won't see it at all in the build instructions but you can still mention it in the instructions.

SmoothieAq, 2021-04-13:
very cool! thank you!

closed 2021-04-13 by SmoothieAq