nophead/NopSCADlib #181: tests/stepper_motors.scad fails on OpenSCAD version 2019.05

cyber-murmel, 2021-05-16:
## Versions

```bash
$ openscad --version
OpenSCAD version 2019.05
$ git reflog
a1e25bb (HEAD -> master, origin/master, origin/HEAD) HEAD@{0}: reset: moving to HEAD
a1e25bb (HEAD -> master, origin/master, origin/HEAD) HEAD@{1}: clone: from https://github.com/nophead/NopSCADlib.git
```

## Steps to reproduce
1. Open tests/stepper_motors.scad in OpenSCAD.
2. Press F5

## Output
```
Loaded design '/home/marble/repos/NopSCADlib/tests/stepper_motors.scad'.
Parsing design (AST generation)...
Used file cache size: 23 files
Compiling design (CSG Tree generation)...
WARNING: Ignoring unknown function 'reverse', in file ../utils/thread.scad, line 101.
WARNING: Ignoring unknown function 'transform_points', in file ../utils/thread.scad, line 105.
WARNING: Ignoring unknown function 'unit', in file ../utils/sweep.scad, line 95.
WARNING: Ignoring unknown function 'unit', in file ../utils/sweep.scad, line 95.
WARNING: Ignoring unknown function 'unit', in file ../utils/sweep.scad, line 95.
WARNING: Ignoring unknown function 'unit', in file ../utils/sweep.scad, line 95.
WARNING: Ignoring unknown function 'unit', in file ../utils/sweep.scad, line 95.
WARNING: Invalid type of parameters for cross(), in file ../utils/sweep.scad, line 41
WARNING: Invalid type of parameters for cross(), in file ../utils/sweep.scad, line 41
WARNING: Invalid type of parameters for cross(), in file ../utils/sweep.scad, line 41
WARNING: Invalid type of parameters for cross(), in file ../utils/sweep.scad, line 41
WARNING: Invalid type of parameters for cross(), in file ../utils/sweep.scad, line 41
WARNING: Invalid type of parameters for cross(), in file ../utils/sweep.scad, line 50
WARNING: Ignoring unknown function 'unit', in file ../utils/sweep.scad, line 51.
ERROR: Assertion '(norm(binormal) > tiny)' failed: "all points are colinear" in file sweep.scad, line 52
TRACE: called by 'fs_frame', in file sweep.scad, line 118.
TRACE: called by 'skin_points', in file thread.scad, line 107.
TRACE: called by 'thread', in file thread.scad, line 189.
TRACE: called by 'female_metric_thread', in file stepper_motor.scad, line 137.
TRACE: called by 'translate', in file stepper_motor.scad, line 136.
TRACE: called by 'for', in file stepper_motor.scad, line 135.
TRACE: called by 'if', in file stepper_motor.scad, line 134.
TRACE: called by 'NEMA', in file stepper_motors.scad, line 27.
TRACE: called by 'rotate', in file stepper_motors.scad, line 26.
TRACE: called by 'let', in file stepper_motors.scad, line 25.
TRACE: called by 'children', in file layout.scad, line 33.
TRACE: called by 'translate', in file layout.scad, line 32.
Compiling design (CSG Products generation)...
Geometries in cache: 414
Geometry cache size in bytes: 54910048
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Compiling design (CSG Products normalization)...
Normalized CSG tree has 0 elements
Compile and preview finished.
Total rendering time: 0 hours, 0 minutes, 0 seconds
```

nophead, 2021-05-16:
It works for me as I run all the tests to make the readme for every change to the library.

`utils/sweep.scad` uses `maths.scad` to get functions like `unit()`. Have you got another `maths.scad` that is somehow being used or is your `maths.scad` corrupted?

My used file cache size is 20, not 23.

nophead, 2021-05-30:
Also NopSCADlib/docs/usage.md says OpenSCAD 2021.01 or later is required.

nophead, 2021-06-01:
Closed because utils/maths.scad uses function literals that are not available in OpenSCAD version 2019.05. Not sure why you didn't see syntax errors in maths.scad rather than it appearing to be ignored.

closed 2021-06-01 by nophead