nophead/NopSCADlib #184: No more updates to NopSCADlib from tomorrow

nophead, 2021-05-31:
Looks like I can't use GitHub from tomorrow as I apparently use a password and can't make personal access tokens work.

I don't enter a password on Windows so I must have stored it somewhere years ago. On Raspbian I have to type my password every time but it doesn't accept a token in its place. The documentation just sends me round in circles so I give up.

martinbudden, 2021-05-31:
That seems a real nuisance. What has caused the password to expire?

nophead, 2021-05-31:
My password hasn't expired but GitHub has deprecated passwords, so every time I use it they tell me it will stop working in June.

martinbudden, 2021-05-31:
Oh dear, I might be facing the same problem.

nophead, 2021-05-31:
That is every time I push from the Windows command line or GUI, where I don't specify a password and can't find out how to change it. I just logged in to my account with my password and it didn't complain.

martinbudden, 2021-05-31:
I see. I think I set up some SSH keys ages ago for pushing from the command line, but I can't really remember what I did.

martinbudden, 2021-05-31:
Doesn't this cover what to do?

https://docs.github.com/en/github/getting-started-with-github/getting-started-with-git/managing-remote-repositories#switching-remote-urls-from-ssh-to-https

nophead, 2021-05-31:
No, because I have never used SSH for git. I am already using https.

martinbudden, 2021-05-31:
The relevant part is:

>The next time you git fetch, git pull, or git push to the remote repository, you'll be asked for your GitHub username and password. When Git prompts you for your password, enter your personal access token (PAT) instead. Password-based authentication for Git is deprecated, and using a PAT is more secure. For more information, see "[Creating a personal access token](https://docs.github.com/en/github/authenticating-to-github/keeping-your-account-and-data-secure/creating-a-personal-access-token)."

Ie create the personal access token and then use it when prompted for a password. Or have you already tried that?

nophead, 2021-05-31:
The problem is l am never asked for a password on Windows. On raspian l am but it won't accept my token.

martinbudden, 2021-05-31:
Have you tried

`git remote set-url --delete ...`

to delete the url and then

`git remote set-url --add ...`

To add it back again?

nophead, 2021-06-01:
```
>git remote set-url --delete origin https://github.com/nophead/NopSCADlib.git
fatal: Will not delete all non-push URLs

>git remote set-url --delete --push origin https://github.com/nophead/NopSCADlib.git
fatal: could not unset 'remote.origin.pushurl'
```

martinbudden, 2021-06-01:
How about a different approach, try making a new clone of the repository and seeing if you can push from there.

nophead, 2021-06-01:
I have just found that the reason I can't find any information on Github or git of how to change the password is because of a huge misunderstanding on my part. Git is set up to use a credential helper on Windows, which is a git component. What I didn't realise is it makes git use the Windows credential helper, which is part of the control panel. So to change the password I just need to use the control panel. The git entries look like this:

[image]

I thought that the git credential helper was the thing that stored the password but it has no GUI, so I couldn't work out how to change it.

closed 2021-06-01 by nophead

martinbudden, 2021-06-01:
Glad you figured this out. And that is good information to have - I wasn't aware of that either.