nophead/NopSCADlib #196: make_all is now not generating anything

niclash, 2021-08-18:
```bash 
niclas@d1:~/dev/3D/DreamPlacer$ make_all.py 
openscad-nightly -D $bom=2 -D $preview=true -o openscad.echo -d bom/bom.deps scad/bom.scad --hardwarnings 
WARNING: Ignoring unknown variable '$_bom' in file ../../NopSCADlib/utils/core/bom.scad, line 45

niclas@d1:~/dev/3D/DreamPlacer$ ls dxfs -l
total 0
```

nophead, 2021-08-18:
The latest version of openscad is broken.

nophead, 2021-08-20:
See https://github.com/openscad/openscad/issues/3781

nophead, 2021-09-12:
It doesn't look like they are going to fix $variables to work like they have for the last 10 years, so I have botched around it. Basically when you use a file then any $variables defined at file level within it are ignored. Only $variables on the call stack are visible.

I also came across a new bug exporting files, which I have also botched around, although I expect it to be fixed. See https://github.com/openscad/openscad/issues/3879

Please can you give it a try.

nophead, 2021-09-13:
Actually although the work arounds stop errors the change to $variables means if you set them at file scope anywhere it will create a bug. Any bespoke ones will become undefined when the framework instantiates STLs and assembly views. Any overrides of $fa, $layer_height, etc will be ignored.

niclash, 2021-09-13:
I looked at it for 15 minutes just now, and realize it will take me a few hours to make the changes to your (better) structure/naming, and don't really have time until coming weekend. Will do it then...

niclash, 2021-09-14:
$job$ stalled yesterday, so I put in the time on it. And after fixing my own issues, it `make_all` works as expected.

niclash, 2021-09-14:
Generated output is in https://github.com/niclash/DreamPlacer

closed 2021-09-14 by niclash