nophead/NopSCADlib #219: Ball bearings should be called cartridge bearings

martinbudden, 2022-02-22:
The things currently called ball bearings should really be called cartridge bearings, see https://duckduckgo.com/?q=cartridge+bearing&atb=v1-1&iax=images&ia=images

I don't think it is worth the bother of changing the names of the code modules and names (eg `module ball_bearing`, `BB608` etc), but I think the name of the vitamins should be changed, eg:

```
vitamin(str("ball_bearing(BB", bb_name(type), "): Cartridge bearing ", bb_name(type), suffix, bb_bore(type), "mm x ", bb_diameter(type), "mm x ", bb_width(type), "mm"));
```

That way the generated BOMs would contain the text "Cartridge bearing".

nophead, 2022-02-22:
I have never heard of cartridge bearings and wouldn't have known what they were. We call them ball bearings in the UK and a search seems to get as many hits: https://duckduckgo.com/?q=ball+bearing&atb=v1-1&iar=images&iax=images&ia=images. 

Here is a company I have bought them from: 
[image] No mention of cartridge bearings.

Are you in the US? Perhaps a UK / US difference in English or perhaps a more technical term versus common language.

martinbudden, 2022-02-22:
I'm English and live in the UK. Before I got into 3D printing, the my main use of bearings was in cycling. The sealed bearing units on bicycles whether for the headset, the bottom bracket or the wheels are pretty well universally called cartridge bearings. See, for example, https://www.bikeradar.com/features/bicycle-bearings-everything-you-need-to-know/

So I guess this is a technical term for bicycles.

martinbudden, 2022-02-22:
OK, withdrawing.

closed 2022-02-22 by martinbudden